dw-0/kiauh #2: A new Klipper webinterface

mbgroot, 2020-07-29:
Maybe you would like to add Mainsail webui to the script ?
https://github.com/meteyou/mainsail

lixxbox, 2020-07-29:
Checkout his dev-2.0 branch 😋

th33xitus, 2020-07-29:
Like lixxbox said, this feature is already available in the dev2.0 branch.
I haven't merged, or rather made it to the new master yet but i will probably do in the next few days.

mbgroot, 2020-07-29:
It's not working yet. - https://github.com/KevinOConnor/klipper/issues/3026#issuecomment-663913808

th33xitus, 2020-07-29:
It is working. But there is a restriction on Mainsail side.
Currently that webinterface only works with a special fork of klipper made by Arksine:
https://github.com/Arksine/klipper/tree/dev-moonraker-testing
**You have to have this version of Klipper active before you can use Mainsail.**

My script actually won't allow you to install Mainsail without being on the correct Klipper Version.
Please use KIAUH for changing the Klipper version, because it will add the remote with a specified name. That's the only way the script can tell if you are on the right version or not, and therefore either decline or allow installation.

As long as all the important parts of dev-moonraker-testing are not merged into the mainline, we have to use the corresponding Klipper fork.

Same goes for people currently using the s-curve smoothing or shaping branch of dmbutyugins fork of Klipper. There will be no way they can use Mainsail as long as the important bits and pieces which are needed for Moonraker are not merged into their forks.

Im sorry, that is nothing i can fix or help with. Thats the way it is for now.... we just need to give the Klipper Devs a little bit more time until they fully implement the Moonraker support. For the meantime there is still the option to use DWC2 or OctoPrint.

mbgroot, 2020-07-29:
Sorry, didn't give the right address in  trusted_clients.

closed 2020-07-29 by mbgroot