dw-0/kiauh #4: Issue listing files on pi zero w / raspbian lite

vladbabii, 2020-08-04:
Issue is on line 507 of kiauh.sh

This produces zero file list
> for script in ${SRCDIR}/kiauh/scripts/*; do . $script; done

This works
> for script in "${SRCDIR}/kiauh/scripts/"*; do . $script; done

Raspbery pi zero, rasbian lite latest version today with all packages updated.
bash version
> root@raspberrypi:~/kiauh# bash --version
> GNU bash, version 5.0.3(1)-release (arm-unknown-linux-gnueabihf)

th33xitus, 2020-08-04:
Thank you for reporting.
Does your {SRCDIR} contains any spaces or why does the `"` fix that issue?

> root@raspberrypi:
Do you try to execute the script as root?

vladbabii, 2020-08-04:
It does not have any spaces, but i have always used "path"/* and it seems to work on windows, linux and mac very well. Yes, i tried at first as root then as user PI. Not sure if it has anything to do with root, but if you run with root you don't see the 'please don't run with root' type message, but a crash of the script on that line.

vladbabii, 2020-08-04:
This was a fresh pi zero w with a fresh raspbian lite install, only ran apt update and apt upgrade -y after first boot (with ssh and wifi configured with the /boot/ssh and /boot/wpa_supplicant.conf - fully headless initial configuration

vladbabii, 2020-08-04:
Also, you can ping me on klipper discord at Q66 if you want to talk there

th33xitus, 2020-08-12:
Fixed.
For now only in dev-2.0.
Will merge into master soon.

-Closed-

closed 2020-08-12 by th33xitus