dw-0/kiauh #6: Feature request: Enable/disable tmpfs on /tmp and /var/log

vladbabii, 2020-08-04:
An option to disable and enable this would be very useful for most users.

These should be added on /etc/fstab to prevent wear on sd card
```
tmpfs           /tmp            tmpfs   defaults,noatime,nodev,nosuid,size=50M 0 0
tmpfs           /var/log        tmpfs   defaults,noatime,nodev,nosuid,size=50M 0 0
```

vladbabii, 2020-08-04:
The memory usage can be kept low by rotating logs with logrotate

create file
```
/etc/logrotate.d/klipper
```

with contents
```
/tmp/*.log
{
  size 1M
  copytruncate
  rotate 0
  su pi pi
}
```

th33xitus, 2020-08-05:
Haven't you had some problems with logs missing and then Nginx not working properly anymore? I think i red something like that on discord? I really don't want to include functions which can cause further, somehwhat unpredictable problems. 

Maybe a better solution would be, to add only klippy.log and moonraker.log to the tmpfs if the user wants? And not the whole tmp-folder? That might work because those files are always created when the services start up, no matter if it already exists or not. And actually it would make sense to me, deleting those files regularly. This klippy.log can get quite bloated.

vladbabii, 2020-08-05:
You can set nginx logs from
> /var/log/ngix/something.log 
to
> /var/log/nginx_something.log

or add an 
> mkdir /var/log/nginx/
in the startup sequence or

add in fstab 
> tmpfs           /var/log/nginx        tmpfs   defaults,noatime,nodev,nosuid,size=10M 0 0

vladbabii, 2020-08-05:
I also added a file in
>  /etc/logrotate.d/klipper

with

```
/tmp/*.log
{
  size 1M
  copytruncate
  rotate 0
  su pi pi
}
```

which makes sure logs aren't filled up

vladbabii, 2020-08-05:
I've also ran printers off read only filesystems except a small partition for gcode files to upload to (or used an usb  stick for that). Those printers had zero filesystem issues and are workhoses. No corruption or broken files. Upgrading is a bit more comples, you need to ssh in, run a command like "root-rw on", then do upgrade, then "root-rw off" the turn it read-only

th33xitus, 2020-08-07:
Okay so i thought a bit about this feature and i came to the conclusion that i won't implement it.

I don't see a benefit for the majority of users at the moment (might be due to own restricted knowledge in that regard). What is the actual benefit regarding the wear of the sd card? Is it safe to always delete the printer symlink in the tmp folder? Does it get recreated everytime the klipper service starts again (i haven't tested, and i don't want to tbh)? Questions i have no answer to, possible causes for some issues i (or rather kiauh) don't want to be the reason for.

And to be honest i don't want my script to write into any important system files, add lines, change lines or delete lines. Those are things i consider to be made by power-users themself if they want. I want to interact with the core linux os as few as possible.

I might add those tips to my readme though if you want, under a section "further tweaks" or so?

vladbabii, 2020-08-11:
Ok.

The benefit is a longer lifetime for the sd card. On all my PIs that run critical stuff the filesystem is fully read only and they write into tmpfs. I've got a couple running for over 4 years under a great number of random power failures and other issues. Cheap sd cards work well for 2-3 months at most, if you write to them a lot of logs and good ones are from 4 to 12 months, but they usually cost 5-10x the price.

closed 2020-08-12 by th33xitus