dw-0/kiauh #7: Trusted clients does not include 10.x.y.z

vladbabii, 2020-08-04:
Not sure if 10.x.y.z/? should be included by default (some routers use that as public ip on some ISPs), but at least a warning in the readme should be added that if your network uses 10.x.y.z you should add it manually in trusted clients.

th33xitus, 2020-08-04:
Im not quite sure if i can or will pay attention to every single edge case regarding possible configurations.
I think the best solution might be, to give the user a choice.
Option 1: Write a default config entry with the currently included trusted clients.
Option 2: Read user input where user can define the trusted client himself, which then gets written to the config file.
Option 3(maybe?): A mix of option 1 and option 2. Default IPs are written anyways, but the user gets the choice to add more himself?

What do you think?

vladbabii, 2020-08-05:
User choice is a good option! I think having an option for user to fill in ip and range would be best. The value could be prefilled based on the network information, search for "192.168.0." or "192.168.1." or "10.0.0." in the network interfaces list and prefill one of the values...

th33xitus, 2020-08-05:
Okay i will take a look into it, but can't say how fast i can/will implement this :)

th33xitus, 2020-08-26:
I was working on a function like bespoken.
This is what i came up with:
[image]

I think this will be a good way?
Additionally, the current IP of the Pi will be written to the config no matter, or rather the IP gets converted to an IP range:
[image]

closed 2020-08-27 by th33xitus