dw-0/kiauh #12: Mainsail install failure

schlick7, 2020-09-09:
This install fails immediately with it complaining about a disable error.

I'm not sure what this code does in install_mainsail.sh, but I had to delete it and then Mainsail installed just fine

  for service in "${disable_service[@]}"
  do
    if [[ $(dpkg-query -f'${Status}' --show $service 2>/dev/null) = *\ installed ]]; then
      status_msg "Disabling $service service ..."
      sudo apt-get disable $service
      ok_msg "$service service disabled!"
    fi
  done

th33xitus, 2020-09-10:
Thank you for reporting that issue.
I work on a fix right now. There went something wrong in the past, i haven't noticed it yet with testing because my system doesn't have haproxy or lighttpd installed. Therefore this step will always be skipped.

th33xitus, 2020-09-11:
This issue should be fixed now and live in master!

closed 2020-09-14 by th33xitus