dw-0/kiauh #21: After installation user pi has no authorization to /dev/ttyUSB0.

anjoku, 2020-10-04:
Hello

many thanks for the installation script.

I installed Mainsail on an Orangepi Lite.

Installed with the user pi. pi is entered as sudoer.

After the installation there is no printer.cfg in /home/pi. I had to copy them via winscp.
is the problem that I'm not using a Raspberry Pi or is it possible to change the script so that a printer.cfg is always available?
After the installation, the user pi has no authorization to /dev/ttyUSB0. The printer is not found.
If I adjust the rights, the printer will be found.

Thank you Andre

th33xitus, 2020-10-04:
Hi, thank you for using KIAUH.

What Linux distro are you using with that OrangePi? RaspberryPi OS?

Usually during the Moonraker installation (at the very beginning actually) there will be checked if a printer.cfg exists:
https://github.com/th33xitus/kiauh/blob/master/scripts/install_moonraker.sh#L28
and
https://github.com/th33xitus/kiauh/blob/master/scripts/functions.sh#L14

The function will read the /etc/default/klipper file for the printer.cfg location. If there isn't a printer.cfg at that location, the script will ask if you want to create a basic printer.cfg. Not really a working config, it's more like just having that file at the right spot to continue later on. I do see a problem here if Klipper was installed as systemd service. I think i need to fix that.

Regarding your permission issue:
The script DOES NOT check for any permissions. I know that might not be comfortable but for now users have to make sure that all permissions are set. I haven't encountered any of those permission issues yet myself.

ciapata, 2020-10-06:
Use this command to grant access to your USB port

`sudo adduser YOUR_USERNAME dialout`

closed 2020-10-21 by anjoku

megablue, 2022-04-05:
ubuntu-20.04 lts also doesn't grant serial port access to a non-root user by default... perhaps it is a good idea to check and automatically add the user to the group

th33xitus, 2022-04-05:
> ubuntu-20.04 lts also doesn't grant serial port access to a non-root user by default... perhaps it is a good idea to check and automatically add the user to the group

Thanks for the info. So basically i should already check during the Klipper installation if the user is in group `dialout` ?
Currently, membership of group `dialout ` is only checked, if you want to flash a MCU.
But now, further thinking of it, it makes sense to always make sure that the user is in that group, so communication with the serialport is possible?

megablue, 2022-04-05:
yup... at first i cant even use klipper because of that, i flashed the firmware manually in the terminal instead of using the script.

th33xitus, 2022-04-05:
I will add a fix for that.