dw-0/kiauh #23: Feature request: option to upload logs

vladbabii, 2020-10-07:
Adding an option to upload logs and provide the url would make things much easier for users.

Many pastebin services allow you to use curl to upload some data, for example
```
curl -s --data 'Hello World!' 'http://paste.c-net.org/'
```
returns an url when run
```
http://paste.c-net.org/SaucersBonjour
```

A warning should be given to the users that logs may contain network information and other private data ( file paths, file names, etc) and be asked to confirm uploading the logs.

I think there are 2 viable options:
1. upload only the last X lines of each file concatenated in a single upload, where X could be set once and remembered or as a hardcoded value initially ( last 1000 lines, for example )
2. upload each file individually ( to try to avoid having a too big upload)

th33xitus, 2020-10-10:
Added with https://github.com/th33xitus/kiauh/commit/29e3d8b3d278c164810f16834dc7f376c67bb2d3

closed 2020-10-10 by vladbabii

b01t, 2020-10-13:
This is awesome!