dw-0/kiauh #27: USB Webcam not working

bruce356, 2020-10-11:
Hi, could you include installation of service/drivers for USB webcams, I have Logitech Webcam C920 but most usb webcams should be the same.
Used "2020-08-20-raspios-buster-armhf-lite.img" on SD card.
In Octopi I normally enter 
/webcam/?action=stream
http://127.0.0.1:8080/?action=snapshot
/usr/bin/ffmpeg
and it works perfectly, but with this installation of octopi webcam does not work.

With the octopi-buster-lite-0.17 image MJPG-Streamer is included.
Would be helpful for inexperienced like me - thanks

th33xitus, 2020-10-11:
No, im sorry.
I don't use a webcam myself and therefore i can't set up a reliable installation routine for that without even being able to test it myself.

bruce356, 2020-10-11:
Thanks for the reply, you created a great script, cadriel/FluiddPI stand alone image has the webcam incorporated so I will use that, its handy to be able to view what the printer is doing from the office computer, printer is located in another room.
Regards - Bruce

closed 2020-10-12 by th33xitus