dw-0/kiauh #29: nginx has only default config

ScotchTape, 2020-10-20:
Hi,

Since I was moving away from DWC2 I decided to install klipper and everything related from scratch. Since I used this script in the past with great success I also installed it from git.
The klipper installation went fine, but moonraker/mainsail not so much:
```
###### Creating klippy.log symlink ...
Symlink created!

Creating moonraker.log symlink ...
Symlink created!

Restarting Moonraker Service ...
Moonraker Service restarted!

Restarting Klipper Service ...
Klipper Service restarted!

Testing API ...

Please wait ...

###### API response from http://192.168.178.244:7125/printer/info :
{"result": {"state_message": "Printer is ready", "klipper_path": "/home/dietpi/klipper", "config_file": "/home/dietpi/klipper_config/printer.cfg", "software_version": "v0.8.0-843-gf8cb1e30", "hostname": "any-klipper", "cpu_info": "4 core ARMv7 Processor rev 5 (v7l)", "state": "ready", "python_path": "/home/dietpi/klippy-env/bin/python", "log_file": "/tmp/klippy.log"}}

Klipper API is working correctly!

Testing Nginx ...

Please wait ...

###### API response from http://192.168.178.244/printer/info :
<html>
<head><title>404 Not Found</title></head>
<body bgcolor="white">
<center><h1>404 Not Found</h1></center>
<hr><center>nginx/1.14.2</center>
</body>
</html>

<!!!!> Nginx is not working correctly!`
```
I poked around a bit, the configs for moonraker are getting installed, but sites-available has only the default and not the mainsail_nginx.cfg out of resources copied and a grep comes back with only one line for DEFAULT_PORT but no sudo cp.

th33xitus, 2020-10-20:
Hi,

I think i know why there is the warning about `Nginx is not working correctly`.
Have you installed a webinterface after installing Moonraker?

In the process of adding Fluidd to the supported interfaces you can install, i removed the option to install Moonraker and Mainsail together after asking a question if you would like to do so (like mentioned here: https://github.com/th33xitus/kiauh/blob/master/docs/changelog.md#2020-10-06).
So for the moment you have to install Moonraker first, and after that you need to install an Interface also.
I think that `<!!!!> Nginx is not working correctly!` comes, because there is no interface installed at the moment the check happens, therefore no proper NGINX configuration for that (like you mentioned, only the default config is available in the sites-available folder) and therefore that error gets thrown.

Please install an interface and report back. If i am not mistaken, things should work out fine and i just need to fix that hopefully soon, to prevent further confusion.

th33xitus, 2020-10-20:
I have pushed a little fix which should adress this issue here as well. It should fix that "false-positive" warning, that NGINX is not working correctly.

Anyways, i would love to hear from you if your issue got resolved after installing an interface and if everything is working now.

ScotchTape, 2020-10-20:
Yes, that fixed it. I was just a bit confused by the 3rd party documentation, which didn't mentioned it ;)

closed 2020-10-20 by ScotchTape