dw-0/kiauh #30: install moonraker incomplete

palouf34, 2020-10-25:
I’m install moonraker on my  thinkclient under debian 10 and the  api test failed

see the traceback

`Traceback (most recent call last):
  File "/home/octo/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1680, in _execute
    result = self.prepare()
  File "/home/octo/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 2435, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found

<!!!!> Klipper API not working correctly!`

thk’s for your help.

th33xitus, 2020-10-25:
Have you updated KIAUH to the lastest commit?

Does the Moonraker service runs fine?

palouf34, 2020-10-25:
I’m just  git the  KIAUH  script

the  moonraker  service  is ok

`
 systemctl status moonraker
 ● moonraker.service - LSB: moonraker daemon
    Loaded: loaded (/etc/init.d/moonraker; generated)
    Active: active (running) since Sun 2020-10-25 14:44:15 CET; 31min ago
      Docs: man:systemd-sysv-generator(8)
   Process: 6434 ExecStart=/etc/init.d/moonraker start (code=exited, status=0/SUCCESS)
     Tasks: 2 (limit: 3717)
    Memory: 16.6M
    CGroup: /system.slice/moonraker.service
            └─6439 /home/octo/moonraker-env/bin/python /home/octo/moonraker/moonraker/moonraker.py

`

th33xitus, 2020-10-25:
Have you already installed a web interface and tried it out? Does Mainsail or Fluidd work?

Maybe the error message was a false positive...

palouf34, 2020-10-25:
Mainsail  or fluid work’s but don’t reconize the  printer config klipper

th33xitus, 2020-10-25:
Ok if Mainsail/Fluidd is working, that was likely a false-positive warning message. What a bummer...i thought i fixed it with the last commits.

Please do a pastbin of your klippy.log and moonraker.log

palouf34, 2020-10-25:
for klippy : [http://paste.c-net.org/CohenArrives]
for moonraker: [ http://paste.c-net.org/HonestCherie]

palouf34, 2020-10-25:
Ok I’m resolw my problem

Debiam usi systemctl and the config file is not update  correctly.
this manaully update the _**/etc/systemd/system/klipper.service**_ with  _**-a /tmp/klippy_uds optio**_n.

and good job after

th33xitus, 2020-10-25:
Sorry for the late response.
Okay now i see where the problem is... i haven't considered the klipper installation as a systemd service. Therefore the script only checks an available init.d service file for the needed parameters.

I will fix that in later commits! Thanks for the heads up!

palouf34, 2020-10-25:
see other path on this  conf file  for path of the printer.cfg

th33xitus, 2020-10-29:
The issue should be fixed. The script can now detect systemd services and patch those files accordingly.

closed 2020-10-29 by th33xitus