dw-0/kiauh #37: Multi instance support

1N4148, 2020-11-11:
Hello,

might be an interesting feature for the future: Multi Instance Support

It's possible to run multiple instances of Klipper/Octoprint/etc on the same Raspberry/etc. to manage more than one connected printer. But it's difficult to setup, different ports, different config-files, different init.d-scripts. 
Maybe Kiahu is extended to manage this from a single point?

th33xitus, 2020-11-12:
Hi,

Yes i was asked several times if kiauh will support multi instances anytime soon or if its planned. At the moment, i am not working on such a functionality but i see that its frequently requested. I have to see if i can manage to wrap my head around a way to provide such functions.

th33xitus, 2021-01-21:
Just a short update.
One might already have noticed, there is a second branch named `kiauh-rework` where i am reworking quite a lot of the script.
Mainly due to implementing multi instance installer capabilities.

It is still not 100% complete but already functional. So if anyone feels adventurous, feel free to checkout `kiauh-rework`. @1N4148

th33xitus, 2021-01-31:
Should be solved with 16e6dde99864bc0c315f483c036938728ae597ba

Closing...

1N4148, 2021-01-31:
Switched to a single Pi per printer so I cannot test at the moment. But I'll prepare a test environment the next days to test your work :)

th33xitus, 2021-01-31:
Awesome :) Feedback and further testing is always welcome! I will reopen this issue here, so we can just continue in here then.

Prutsium, 2021-03-17:
Found your tool by accident and looks great as i want to run multiple printers.
But i have the FluiddPi image and seems as your installer did not install everything it wont work to make multiple instances of Moonraker / Klipper.

Any idea how to proceed?

th33xitus, 2021-03-17:
@Prutsium If you are starting off a fresh FluiddPi image you already have Klipper and Moonraker installed.
You need to uninstall the pre-installed installations first, then run the KIAUH Installer for Klipper again. You will then get asked how many instances you want to set up.

My own tests with MainsailOS and FluiddPi worked.

Prutsium, 2021-03-17:
> @Prutsium If you are starting off a fresh FluiddPi image you already have Klipper and Moonraker installed.
> You need to uninstall the pre-installed installations first, then run the KIAUH Installer [...]

Sorry to tag you again but ok i have 2 x Klipper, 2 x Moonraker and both api's are up.
Now i want to add the 2nd printer to Fluidd but how do i enter it? add printer and IP:port seems not be the way.
[image]

teikjoon, 2021-09-27:
> Sorry to tag you again but ok i have 2 x Klipper, 2 x Moonraker and both api's are up.
> Now i want to add the 2nd printer to Fluidd but how do i enter it? add printer and IP:port seems not be the [...]

I solved this but using the IP address of the host (e.g. 192.168.0.200:7126) for the 2nd instance of moonraker (and by extension klipper)

teikjoon, 2021-09-27:
Not sure if I should necro this old thread instead of opening a new one, but it seems the multiple instance is working...I have 2x klipper/moonraker running on one host and everything works, except for the 'restart' functions.

This is probably since fluidd is trying to restart the service 'klipper' instead of 'klipper-1' or 'klipper-2' that is set up by KIAUH.

Is there any way we can have one more configuration variable, i.e. a sequence # or simply some variable which we can define the service name (e.g. klipper-1, moonraker-1) so that the 'restart service' button works?

[image]

Thanks.

th33xitus, 2021-09-27:
@teikjoon That is something the webinterface has to support. If you take a look at Mainsail for example, you will see that it's already possible there it and will show the additional services. Fluidd doesn't support that yet and needs an update to do so.

teikjoon, 2021-09-28:
> @teikjoon That is something the webinterface has to support. If you take a look at Mainsail for example, you will see that it's already possible there and it will show the additional services. Fluidd doesn't support that yet and needs an update to do so.

Thanks for quick reply @th33xitus - I guess I'll have to go log an issue with Fluidd then :)

susisstrolch, 2021-09-28:
Simple fix:
create /etc/systemd/system/klipper.service
```
[Unit]
Description=Bundle all klipper-x services
After=network.service

[Service]
Type=oneshot
ExecStart=/bin/true
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```
create /etc/systemd/system/moonraker.service
```
[Unit]
Description=Bundle all moonraker-x services
After=network.service

[Service]
Type=oneshot
ExecStart=/bin/true
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```
in klipper-*.service replace the Unit and Install section...
```
[Unit]
Description=Starts klipper instance 1 on startup
After=klipper.service
PartOf=klipper.service

[Install]
WantedBy=klipper.service
```
in moonraker-*.service replace the Unit and Install section...
```
[Unit]
Description=Starts Moonraker xxx on startup
After=moonraker.service
PartOf=moonraker.service

[Install]
WantedBy=moonraker.service
```
execute
```
sudo systemctl daemon-reload
sudo systemctl enable klipper.service moonraker.service
sudo systemctl start klipper.service moonraker.service
```
Now "sudo systemctl restart klipper" restarts all klipper instances w/o any change to the frontend.