dw-0/kiauh #39: Issue with python check

melvinisken, 2020-11-19:
I'm using an older install of an RPi 3 with Raspbian 9.13 stretch, but I think this should not be related to that?

I get this error every time I try to install Moonraker:
` 
 Python 3.7 or above required!           
 Please upgrade your Python version first.           
 `

I installed Python 3.9 and if I run python3 --version on command line, I get "Python 3.9.0" as a result:
`$ python3 --version`
`$ Python 3.9.0`
Shouldn't the script recognize that the version is above 3.7? Or is the result on buster printed differently (I have no device to test right now)?

th33xitus, 2020-11-19:
Hi,
I upgraded my Python version from 3.7 to 3.9 just to test this issue you are reporting but i can't reproduce it...
The check which is done is pretty simple...
```
if [ $(python3 --version | cut -d"." -f2) -ge 7 ]; then
    <do stuff>
else
    <error message>
fi
```

So as soon as `[ $(python3 --version | cut -d"." -f2) -ge 7 ]` evaluates to true, which is does when the second number in the python 3 version number is greater or equal to 7, the moonraker installation will start.

So something like `if [ $(python3 --version | cut -d"." -f2) -ge 7 ]; then echo "true"; fi` will still print true to my console now.

melvinisken, 2020-11-20:
I just tested, using `if [ $(python3 --version | cut -d"." -f2) -ge 7 ]; then echo "true"; fi` on command line does result in "true", but if I select 1 - install -> 2 - moonraker it results in the same error message. Weird.

th33xitus, 2020-11-20:
I have to do some rework on the moonraker installer anyways. I will rewrite the python version check a little bit, make it its own function and then we can try again if that works out. At the moment i really have no clue why this is not working for you.

melvinisken, 2020-11-20:
In the meantime I just wiped the installation and set up a fresh rasbian lite image and here it works without issues. I don't know what the problem could have been. 
Thank you very much anyways!

closed 2020-11-20 by melvinisken

th33xitus, 2020-11-20:
Interesting. So i guess it might be due to the manual update of python3 then?
I also encountered issues when trying to install moonraker with a manually updated python3 version.
Setting up the virtual environment failed then iirc.

But im happy to hear, that everything works now for you! Cheers.