dw-0/kiauh #41: Self update option?

th3fallen, 2020-11-23:
Any thoughts for the ability to self update kiauh along with the other stuff in the update menu?

th33xitus, 2020-11-23:
There is actually an update notification in bright yellow in the main menu if there were new commits to KIAUH.
[image]
Is that not okay? Not sure if there needs to be a separate entry in the update menu then?

th3fallen, 2020-11-23:
ah that's totally fine i ddint realize it did that. thanks!

closed 2020-11-23 by th3fallen

th33xitus, 2020-11-23:
I see, actually you are not the first person not really noticing that notification. There were several people overlooking it. :D 
Maaaaybe i will think about a different, more prominent way to display a possible update, but for now i would stick to that notification :)