dw-0/kiauh #42: Problems updating scurve-shaping branch

1N4148, 2020-11-24:
Get this when trying to update scurve-shaping branch:

###### Checking out dmbutyugin
error: pathspec 'dmbutyugin' did not match any file(s) known to git

Installed Klipper, Moonraker & Fluidd with Kiahu, switched also from master to scurve-shaping, but now I cannot update with Kiahu.

Manual update is working:
>git fetch dmbutyugin
>git checkout dmbutyugin/scurve-shaping

th33xitus, 2020-11-24:
Thanks for the report, i will have a look at it tomorrow.

th33xitus, 2020-11-26:
I was able to reprocude the error. I think i solved the issue, it was a typo in the update function.
Can you please update KIAUH and try again please? In my tests, everything worked fine again now.

1N4148, 2020-11-26:
Looks good, but meanwhile I've updated the branch manually.

But doing an update on the same branch doesn't show the error message anymore :)

commit 42a5a0b6 (2020-11-26): fix: fix typo, should fix #42
closed 2020-11-26 by th33xitus