dw-0/kiauh #43: Install fails if "dbus" is not installed.

edocod1, 2020-11-26:
Hello!
I would suggest to add "dbus" between the requirements to install using apt since on DietPi the install has been failing since by default it misses dbus.

The error i get is "Failed to create bus connection: No such file or directory" whenever the scripts wants to check if the services exist.

th33xitus, 2020-11-26:
Between the requirements to install what component?
You need to be a bit more specific there.

edocod1, 2020-11-26:
Hello, sorry!

This happens when trying to install klipper as a systemd unit, so i guess the requirement would be needed there.

th33xitus, 2020-11-26:
I just pushed the fix, can you please check and report back if it works now?
Remember tu update KIAUH first :)

LuciusDante, 2020-11-30:
weird thing is I perform an update for fluid, mainsail and it gives me something like wget fault later my fluid interface broken, Ithink owner of the fluid change something

th33xitus, 2020-12-01:
@LuciusDante i don't understand. what you say doesn't really seem to be related to the actual issue/topic here?

LuciusDante, 2020-12-01:
Sorry my bad

beikeland, 2021-06-16:
This seems to be an issue still, DietPi distro does not have dbus package installed by default and this script relies on it.

I found this issue after struggling to have kiauh install moonraker. `sudo apt install dbus` resolves it. Prior to installing dbus kiauh would promt me to install 0 moonraker instances and the install would be incomplete.

beikeland, 2021-06-16:
> Did you install Klipper with KIAUH too, before you tried installing Moonraker with it?

yes, i did and Klipper succeeded, but I still had to install dbus with apt before moonraker worked somehow. can repeat the experiment next week to be sure.

th33xitus, 2021-06-18:
@beikeland I installed myself a second VM with DietPi on it, luckily they provide a VMware image themself.
So i could directly test KIAUH on a DietPi distro.
Unfortunably i wasn't able to reproduce your issue. dbus got installed right away during the Klipper installation without any problem at all.

closed 2021-06-29 by th33xitus

borgaspower, 2021-11-04:
> @beikeland I installed myself a second VM with DietPi on it, luckily they provide a VMware image themself. So i could directly test KIAUH on a DietPi distro. Unfortunably i wasn't able to reproduce [...]

Just install DBUS and reboot

commit dde5812c (2020-11-26): fix: add check for dbus dependency (fix #43)