dw-0/kiauh #46: Might be a stupid question but...

Jord4n, 2020-11-27:
So does this actually update and flash my board when doing the update feature?  Or do I still need to build and manually flash the board via "make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0"?

th33xitus, 2020-11-28:
You can build and flash the firmware with the script as well.
First do the klipper update through the update menu as usual.
After updating go back into the main menu, then switch to the advanced menu and select `Build + Flash MCU`.

Jord4n, 2020-11-28:
Ok thank you, I guess that does make sense.  I was confuse as when I see "update" I figured it was already flashing the board for me in the background.  To further confuse things I misunderstood "remote" version being the board itself so I always thought when the versions match I was just good to go.

Cheers!
J

th33xitus, 2020-11-28:
No no :) You need to explicitly tell the script to flash.
And remote always means the most current available version on the internet (github).

closed 2020-11-28 by th33xitus