dw-0/kiauh #48: unable to access 'https://github.com/th33xitus/kiauh.git/': server certificate verification failed

dalswede, 2020-12-09:
Hi,

I have an issue with accessing Kiauh on rpi4. It had been working fine for weeks, however when I tried to connect today to check for updates, it comes up with the following error :

`$ git clone https://github.com/th33xitus/kiauh.git
Cloning into 'kiauh'...
fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': server certificate verification failed. CAfile: none CRLfile: none`

I attempted to re-instal Kiauh and it comes up with same error message. I checked to make sure git was up to date. Any advice on what's causing this?

Many thanks

KwadFan, 2020-12-09:
I think this based on your ca-certificate package. Try apt update and look for possible update. Otherwise try updating certificates [Manpage/update-ca-certificates](https://manpages.debian.org/buster/ca-certificates/update-ca-certificates.8.en.html)

dalswede, 2020-12-10:
Thanks for the feedback and advice. It turned out I had a corrupt installation on my USB stick that I am using on rpi4. Kept getting error messages about perl missing when trying to run apt update. Just trying to re-image now and set it all back up again.

dalswede, 2020-12-10:
issue was a corrupt instalation

closed 2020-12-10 by dalswede