dw-0/kiauh #49: Fail when running update option from main menu

numanair, 2020-12-11:
Fresh install of kiauh, so it is already up-to-date. Selecting update (2) from the main kiauh menu results in the following error:
> fatal: '4718f39b2f8ef326dc671aa936c509e98fd6cf9f' does not appear to be a git repository
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.

Let me know if you need more info or logs.

th33xitus, 2020-12-11:
The error message is corresponding to this commit
https://github.com/KevinOConnor/klipper/commit/4718f39b2f8ef326dc671aa936c509e98fd6cf9f
That is from march 2019. You haven't updated since?

I assume you have fiddled arround with your local klipper git repository in the past?

At the moment my best solition for you would be, to remove klipper and re-install it again.
It should be even enough to quit KIAUH and delete the klipper folder manually with:
`cd ~ && rm -rf klipper` , this command will remove the klipper folder from your users home directory.
You can then start KIAUH again and the update menu should work. Since the klipper service is still installed, you can issue a simple klipper update then.

BUT.... if you really haven't updated since march 2019, i would even consider removing klipper as a whole (remove menu from KIAUH) and re-install klipper as a whole, including its service. At least you then can be sure, to have a rather clean and fresh klipper installation.

numanair, 2020-12-12:
Oh, yeah I am on an old Klipper version on that install. I thought the update option was just for updating the kiauh tool, but I guess I misunderstood. My other printer is running a recent build, so I really should be messing around with that one instead.

closed 2020-12-13 by numanair