dw-0/kiauh #56: line 363: /home/pi/kiauh/kiauh.ini: Permission denied

WhatTakWhat, 2021-01-11:
pi@raspberrypi:~/kiauh $ ./kiauh.sh 
/home/pi/kiauh/scripts/functions.sh: line 363: /home/pi/kiauh/kiauh.ini: Permission denied

on raspberry PI 4 with Raspberry Pi OS lite.

Please help, what’s happen?

th33xitus, 2021-01-11:
Hi, can you please post the output from the following command?
`cd ~ && ls -l ~/kiauh/kiauh.ini`
This command will show the permission settings for the kiauh.ini file.
Without knowing what they are set to, i can't really help.

WhatTakWhat, 2021-01-11:
> Hi, can you please post the output from the following command?
> 
> `cd ~ && ls -l ~/kiauh/kiauh.ini`
> 
> This command will show the permission settings for the kiauh.ini file.
> 
> Without knowing what they are set to, i can't really help.

Hello!

Sorry to bother you, when I came to home - reinstall Pi OS lite and bug disappear.

But before I do this - it was completely repeatable error. Sorry, I don't save any data.

Thank you for your scripts and fast answer. Hope today I will finally use klipper+dwc.

closed 2021-01-14 by th33xitus

WhatTakWhat, 2021-01-14:
Hello!

BTW - there was no .ini file, when I create this issue, but after re-installing on clean RPI OS lite everything work like expected

th33xitus, 2021-01-14:
Hi, thanks for the feedback. But sadly i can't give any explanation what could have caused that.
Glad to hear that it works now.