dw-0/kiauh #57: Add update_manager to moonraker.conf for fluidd

noobydp, 2021-01-14:
Fluidd 1.2.0 added support for auto updates via the UI.
If you add the section below to the moonraker.conf file fluidd's update manager will start working.

https://github.com/cadriel/fluidd/releases/tag/v1.2.0

```
[update_manager]
client_repo: cadriel/fluidd
client_path: /home/pi/fluidd
```

th33xitus, 2021-01-14:
I will not add a UI specific entry there since its not only Fluidd that supports the Moonraker update manager but Mainsail as well. So i rather think about giving the user a choice. I'll have a look.

dewi-ny-je, 2021-01-28:
The syntax seems wrong. I think it should be

```
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd
```

th33xitus, 2021-01-28:
@dewi-ny-je the syntax got updated a few days ago. at the time this issue was opened, syntax was fine :)

i hope i will be able to merge soon, some more tests need to be done, then the big kiauh rework is done.

th33xitus, 2021-01-31:
Should be solved with 16e6dde99864bc0c315f483c036938728ae597ba

Closing...

closed 2021-01-31 by th33xitus