dw-0/kiauh #59: KlipperScreen Installation nothing happens?

nicedevil007, 2021-01-20:
Heya, I installed KlipperScreen with root before.
Now I removed the service and tryed to install it with your script.

Nothing more than this is happening/showing to me ?

[image]

th33xitus, 2021-01-20:
@jordanruthe Any idea on that issue?

jordanruthe, 2021-01-20:
@th33xitus https://github.com/th33xitus/kiauh/pull/60 This should fix that issue. I had the same problem and the patch worked afterwards

th33xitus, 2021-01-20:
Cool, merged. Thanks for the quick response!

@nicedevil007 would be nice if you can give feedback if this patch solves the issue.

nicedevil007, 2021-01-20:
Seems to work :) thank you, was able to update as well now!

[image]

closed 2021-01-21 by th33xitus