dw-0/kiauh #62: update RPi OS

ronald5555, 2021-01-21:
Why not add the an update section for the RPi OS software ?

something like:

sudo apt update
sudo apt full-upgrade

th33xitus, 2021-01-21:
I think you are better of with simply defining an alias if you use those commands often.
It doesn't make much sense to my why one would start kiauh first, then navigate to the update menu just to basically execute 2 commands which he could have already executed instead of starting kiauh.

This will add an alias with the name "upgrade":
`cd ~ && echo "alias upgrade='sudo apt update && sudo apt full-upgrade'" >> ~/.bash_aliases && exec bashrc`

From then on you can just SSH into the Pi, type "upgrade" and both commands you wished i should add an update section for will get executed.

Easier, faster... and especially a lot less work for me for only executing 2 commands in the end.

ronald5555, 2021-01-22:
Hello @th33xitus 

I agree with your comment.

In the same idea, could it be possible to backup the complete Klipper_config folder ?
This folder holds all the cfg files and menu files ??

Ronald

th33xitus, 2021-01-22:
Hi,

such feature is already in the making. But not on the master branch. I am working on a quite big rework of the script (as you can see by the commits here https://github.com/th33xitus/kiauh/commits/kiauh-rework) which goal it is to make almost every installable part multi-instance capable (that means more than one instance of klipper/moonraker/fluidd/octoprint/dwc2 on one single raspberr pi). In that branch, the feature already exists. I hope i can merge this to the master branch in the near future. At the moment i need some more time and maybe some more testers who give feedback on that or find bugs.

But yeah tl;dr: Will be there (hopefully) soon:tm:  :)

th33xitus, 2021-01-31:
Should be solved with 16e6dde99864bc0c315f483c036938728ae597ba

Closing...

closed 2021-01-31 by th33xitus