dw-0/kiauh #63: Updating KIAUH Script fails

nicedevil007, 2021-01-23:
Heya just noticed that you updated the KIAUH script and I should type "update" in the menu to update it.

```
/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Updating KIAUH ...
Updating 43a724b..bb0eb69
error: Your local changes to the following files would be overwritten by merge:
        scripts/install_klipperscreen.sh
Please commit your changes or stash them before you merge.
Aborting

```

Leaded to this.

nicedevil007, 2021-01-23:
Fixed it by myself with deletion of the old install_klipperscreen.sh, but this wasn't edited by myself before o_O

th33xitus, 2021-01-23:
The error message states, that you have made changes to the file `install_klipperscreen.sh`.
If you modify a script yourself, it can't be updated anymore that easily.

KwadFan, 2021-01-23:
I am facing the same Issue. I haven't made any changes accordingly. My Solution was a bit rough but it worked. 
```git reset --hard``` 
Cheers

th33xitus, 2021-01-23:
Okay that is weird.
The file got updated 2 days ago, but that shouldn't create any issue. I have no idea why git then complains about some (rather) non-existent local changes. I try and investigate this issue.

th33xitus, 2021-01-23:
@nicedevil007 so you updated your kiauh from 43a724b80eae386a4c17b2d8582f4f11bbf9b9e6 . that was a point in history where the install_klipperscreen.sh fix wasn't applied yet. Are you sure you haven't modified the file yourself before.
In https://github.com/th33xitus/kiauh/issues/59 you said, you updated kiauh after that patch from jordan was provided and i merged it. But that would mean, your kiauh would be already PAST commit 43a724b80eae386a4c17b2d8582f4f11bbf9b9e6.

@KwadFan From which commit did you make the update? The same as nicedevil?

i just updated my own kiauh on my production pi...
[image]

went through without an issue. and i had that klipperscreen fix already applied as well as you can see because the update startet from 8bf20aa05912524beaff2560981584d31c5ec7e4

KwadFan, 2021-01-23:
It was commit f1289504

th33xitus, 2021-01-23:
@KwadFan okay so you haven't updated for a longer time then. That makes troubleshooting a bit more complicated.

Especially now that i got a screenshot from someone who updated successfully exactly after 43a724b80eae386a4c17b2d8582f4f11bbf9b9e6
[image]
and someone who updates successfully after a44ae4c2b993f7d341eae8e9435dd6e3f8adb7b4
[image]

At the moment i don't have an idea where to look for the issue? It seems very random to me

nicedevil007, 2021-01-23:
> @nicedevil007 so you updated your kiauh from [43a724b](https://github.com/th33xitus/kiauh/commit/43a724b80eae386a4c17b2d8582f4f11bbf9b9e6) . that was a point in history where the install_klipperscreen.sh fix wasn't applied yet. Are you sure you haven't modified the file yourself before.
> In #59 you said, you updated kiauh after that patch from jordan was provided and i merged it. But that would mean, your kiauh would be already PAST commit [43a724b](https://github.com/th33xitus/kiauh/commit/43a724b80eae386a4c17b2d8582f4f11bbf9b9e6).
> 
> @KwadFan From which commit did you make the update? The same as nicedevil?
> 
> i just updated my own kiauh on my production pi...
> [image]
> 
> went through without an issue. and i had that klipperscreen fix already applied as well as you can see because the update startet from [8bf20aa](https://github.com/th33xitus/kiauh/commit/8bf20aa05912524beaff2560981584d31c5ec7e4)

I'm pretty sure that I didn't do that even by accident... can't remember that I ever went into the scripts folder in the kiauh directory...

KwadFan, 2021-01-23:
@th33xitus
Yes, not running latest and greatest brings sometimes major issue to the table.
Updates haven get the Attention they should have on my printer :)

In my opinion, I would brute force the update like 
```git reset --hard```
because in my mind KIAUH shouldnt be modified, unless you know what are you doing.
And at this point you should know to prevent such Errors or even to deal with them.
Its more curious for lazy people or peoples without the technical background in mind of Linux Administration and Shell scripting.
So these last named should live with the circumstance that you made the Update rules. Period.
So, if you ask me, brute force the Update to prevent such randomly happening Issues.
I think this issue could be closed for now.
Regards.

th33xitus, 2021-01-23:
> So, if you ask me, brute force the Update to prevent such randomly happening Issues.

You idea is good. I think i will implement such brute force reset. I agree with you in every point you mention.

th33xitus, 2021-01-23:
https://github.com/th33xitus/kiauh/commit/30ca3967725bc98d13dfb80be26493b3f16f5e90 should solve this issue from now on.
Although i still would like to know why that occurred suddenly...

closed 2021-01-23 by th33xitus