dw-0/kiauh #65: cannot install in another folder

tjengbudi, 2021-01-29:
looks like it cannot change to another folder than home directory.
is there any hard coded home path besides in kiauh.sh?

i try to install in folder "installer" cause it will make my folder clean and not make confused.

th33xitus, 2021-01-29:
What do you mean by install?
It depends where you clone KIAUH into. You aren't forced to clone KIAUH in the home directory.
It should work in any directory or subdirectory you clone it into.
Or do you experience issues? If yes, please provide error messages.

tjengbudi, 2021-01-30:
yes i mean clone into another directory than home. i can clone the repository in another folder. but the script cannot work. cause it seeking in HOME directory.

i try with create folder ".kiauh" and failed. and then trying again with folder "installer"

the error after cloning and execute the script is like below
```
/home/octoprint/installer/kiauh/scripts/status.sh: line 3: cd: /home/octoprint/kiauh: No such file or directory
octoprint@HomeAssistant:~/installer/kiauh$

```

th33xitus, 2021-01-31:
You are right. There is a hardcoded ${HOME} in the code. Thanks for pointing that out, i will fix the issue and check the script if there are any other spots where i made the same mistake.

tjengbudi, 2021-01-31:
emm it is a little weird. i run the kiauh.sh in the separeted folder (let's say in the "installer" folder) and it is force to create the installation in the home root directory instead in the installer directory?

th33xitus, 2021-01-31:
What installation are you talking about now exactly?
Klipper?

tjengbudi, 2021-01-31:
sorry to make you confused about this...

i will told you what i have do 
1. i create folder "simple"
2. make simple as active directory with "cd simple"
3. do "git clone https://github.com/th33xitus/kiauh.git"
4. and "./kiauh/kiauh.sh"

the installation of the klipper, moonraker, and fluidd (cause i install fluidd only) is in the home directory. (/home/octoprint/ in my case), when it supposed need to install in the simple directory (/home/octoprint/simple) 

that's why i ask you, are this script force to install in the home directory? i already tried your newest project. it still force install the klipper, moonraker and the fluid in the home directory.

sorry for my bad english, install might means where you save the klipper, moonraker and fluid code

th33xitus, 2021-01-31:
Now i get what you mean. 😄 

And yes, klipper, moonraker and fluidd (and everything else) gets installed to the home directory. That is hardcoded and that won't get changed, i'm sorry.

closed 2021-02-10 by th33xitus

kjkent, 2023-12-12:
As a feature request and alternative to allowing arbitrary install locations, perhaps KIAUH could respect the XDG Base Directory Specification if the requisite env vars were set. I may fork KIAUH to try to implement this, and would be happy to submit a PR if @dw-0 was open to it.

gusarg81, 2024-11-30:
Hi, i know this is old, but would be nice if the installer let user to chose where to install all the apps. I currently use this in a Rock64 SBC, which I use it at the same for other tasks/apps/services. So having all those folders in my HOME is kind "annoying".

commit dc08d8e7 (2021-01-31): fix: should fix #65