dw-0/kiauh #67: After uninstalling nginx + remove fluidd + reinstalling fluidd => nginx wont work

nicedevil007, 2021-02-21:
As the title says the installation in the order of the title nginx won't start anymore.

Output of `systemctl status` is like this:

```
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-02-21 21:46:40 GMT; 21s ago
     Docs: man:nginx(8)
  Process: 1427 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)

Feb 21 21:46:40 fluiddpi systemd[1]: Starting A high performance web server and a reverse proxy server...
Feb 21 21:46:40 fluiddpi nginx[1427]: nginx: [emerg] host not found in upstream "apiserver" in /etc/nginx/sites-enabled/fluidd:38
Feb 21 21:46:40 fluiddpi nginx[1427]: nginx: configuration file /etc/nginx/nginx.conf test failed
Feb 21 21:46:40 fluiddpi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Feb 21 21:46:40 fluiddpi systemd[1]: nginx.service: Failed with result 'exit-code'.
Feb 21 21:46:40 fluiddpi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
```

And this is what the nginx.conf looks like

```
pi@fluiddpi:~ $ cat /etc/nginx/nginx.conf
user www-data;
worker_processes auto;
pid /run/nginx.pid;
include /etc/nginx/modules-enabled/*.conf;

events {
        worker_connections 768;
        # multi_accept on;
}

http {

        ##
        # Basic Settings
        ##

        sendfile on;
        tcp_nopush on;
        tcp_nodelay on;
        keepalive_timeout 65;
        types_hash_max_size 2048;
        # server_tokens off;

        # server_names_hash_bucket_size 64;
        # server_name_in_redirect off;

        include /etc/nginx/mime.types;
        default_type application/octet-stream;

        ##
        # SSL Settings
        ##

        ssl_protocols TLSv1 TLSv1.1 TLSv1.2; # Dropping SSLv3, ref: POODLE
        ssl_prefer_server_ciphers on;

        ##
        # Logging Settings
        ##

        access_log /var/log/nginx/access.log;
        error_log /var/log/nginx/error.log;

        ##
        # Gzip Settings
        ##

        gzip on;

        # gzip_vary on;
        # gzip_proxied any;
        # gzip_comp_level 6;
        # gzip_buffers 16 8k;
        # gzip_http_version 1.1;
        # gzip_types text/plain text/css application/json application/javascript text/xml application/xml application/xml+rss text/javascript;

        ##
        # Virtual Host Configs
        ##

        include /etc/nginx/conf.d/*.conf;
        include /etc/nginx/sites-enabled/*;
}

#mail {
#       # See sample authentication script at:
#       # http://wiki.nginx.org/ImapAuthenticateWithApachePhpScript
#
#       # auth_http localhost/auth.php;
#       # pop3_capabilities "TOP" "USER";
#       # imap_capabilities "IMAP4rev1" "UIDPLUS";
#
#       server {
#               listen     localhost:110;
#               protocol   pop3;
#               proxy      on;
#       }
#
#       server {
#               listen     localhost:143;
#               protocol   imap;
#               proxy      on;
#       }
#}
```

th33xitus, 2021-02-22:
With purging the nginx installation you also removed the nginx configuration which gets installed during the moonraker installation.
That one is missing now. Please reinstall moonraker.

nicedevil007, 2021-02-22:
Thank you for clearing up, that helped :)

closed 2021-02-22 by nicedevil007