dw-0/kiauh #69: Moonraker fails to start after upgrade

mjoconr, 2021-03-07:
Hi
I'm getting an error after upgrading Moonraker v0.4.1-4
No matter what I do Moonraker is failing to find the printer config
[moonraker.log](https://github.com/th33xitus/kiauh/files/6096906/moonraker.log)

Any ideas ?

mjoconr, 2021-03-07:
Ok, well it was not actually a moonraker error is was the start script for klipper for systemd. I did not edit this so the issue must have been in kiaud when it upgraded klipper.

th33xitus, 2021-03-07:
```
Unable to open config file /home/pi/klipper_config/home/pi/klipper_config/printer.cfg
```
This is impossible to happen with only upgrading Moonraker or Klipper because no config or no service will be rewritten/modified during upgrading. You did something else, to mess this up somehow.

Do you have it solved by now? Or do you still need help?

closed 2021-03-09 by th33xitus