dw-0/kiauh #70: Port changing from Fluidd to Octoprint.

l-Kage-l, 2021-03-11:
So basically, i wanted to intall both Fluidd and Octoprint interfaces. I had OctoPi installed before, installed KIAUH, had to remove Klipper and Octoprint and reinstall them. Did that too.

Then i installed Fluidd first, afterwards Mainsail, afterwards Octoprint. All is done and well, except i've got a problem in Octoprint.

I cannot see the webcam. The relative path, /webcam/?action=stream doesnt work anymore because the Octoprint was installed on a 5000 port instead of the usual 80. Fluidd is on the 80 port now and the camera works on Fluidd with the relative path, but on Octoprint it doesnt because it cannot acces the 8080 port that is assigned to the camera. 

Now i see 2 solutions: Reintall Octoprint and Fluidd, with Octoprint first and Fluidd after and Octorpint should get the 80 port assigned, but this is bothersome because i would have to resintall all of Octoprint's plugins.

The second one being just changing the ports from Fluidd to Octoprint, but i dont know how to do that and google doesnt really help.

Besides the camera issue, maybe this could be implemented in the next KIAUH release, this being the fact that you could choose the port on which to install what interface. If anyone knows how to change the port, it'd be much appreciated

hawktrainer2222, 2021-03-15:
> So basically, i wanted to intall both Fluidd and Octoprint interfaces. I had OctoPi installed before, installed KIAUH, had to remove Klipper and Octoprint and reinstall them. Did that too.
> 
> Then i installed Fluidd first, afterwards Mainsail, afterwards Octoprint. All is done and well, except i've got a problem in Octoprint.
> 
> I cannot see the webcam. The relative path, /webcam/?action=stream doesnt work anymore because the Octoprint was installed on a 5000 port instead of the usual 80. Fluidd is on the 80 port now and the camera works on Fluidd with the relative path, but on Octoprint it doesnt because it cannot acces the 8080 port that is assigned to the camera.
> 
> Now i see 2 solutions: Reintall Octoprint and Fluidd, with Octoprint first and Fluidd after and Octorpint should get the 80 port assigned, but this is bothersome because i would have to resintall all of Octoprint's plugins.
> 
> The second one being just changing the ports from Fluidd to Octoprint, but i dont know how to do that and google doesnt really help.
> 
> Besides the camera issue, maybe this could be implemented in the next KIAUH release, this being the fact that you could choose the port on which to install what interface. If anyone knows how to change the port, it'd be much appreciated
> 
> Edit: Oh and btw, the ADXL345 doesnt work anymore after the switch to KIAUH. Keeps spitting out this: Invalid adxl345 id (got 0 vs e5)

I was having this issue also and resolved as described below (you will already know a lot of this, but I'm just copying/pasting from a comment i left on a facebook group):

because Kiauh installs octoprint at the 5000 port, in order to run octoprint I have to enter the URL with ":5000" added. for example if the hostname is "octopi" you would enter "octopi.local:5000/" to get to your octoprint UI. If using an IP it's the same but replace the hostname.local with your pi IP address.

you can edit your hostname by ssh into your pi then run "sudo raspi-config" then under "system options" > "hostname"

If you also have fluidd or mainsail installed they get installed by kiauh to the default :80 port, in which case the ":80" is not needed.

If you have Fluidd AND Mainsail installed, Kiauh should have asked you to install one of them at a different port.
For example, if your hostname on the PI is "octopi" and in Kiauh you installed mainsail first, then fluidd at port 81, then octoprint you would access the various UI's as follows:
Mainsail: "http://octopi.local/"
Fluidd: "http://octopi.local:81/"
Octoprint: "http://octopi.local:5000/"
webcam stream: add first via mainsail, then to octoprint or fluidd with "http://octopi.local/webcam/?action=stream"

hope that helps!

th33xitus, 2021-03-16:
> Edit: Oh and btw, the ADXL345 doesnt work anymore after the switch to KIAUH. Keeps spitting out this: Invalid adxl345 id (got 0 vs e5)

That sounds somewhat unlikely although i can't verify it.
KIAUH does not change anything on the Klipper Code.

closed 2021-03-19 by th33xitus