dw-0/kiauh #73: No mjpg streamer when install via kiauh

jonathanoeijoeng, 2021-03-19:
Could you help to add mjpg streamer in kiauh?
I have installed klipper, moonraker and fluidd via kiauh and found there is no mjpg streamer in my installation comparing when i install via fluiddpi image.

NovaViper, 2021-03-21:
Can confirm the same issue here on my Raspberry pi 4

th33xitus, 2021-03-25:
Fixed with 104089e :)

closed 2021-03-25 by th33xitus

persano, 2022-11-25:
Did not want to open a new issue, but mjpg streamer is broken or disabled in 5.0.0 

" Function currently disabled! Sorry! " is the error

Creo2005, 2022-11-25:
export KIAUH_SRCDIR=${HOME}/kiauh
export PRINTER_DATA=${HOME}/printer_data
export KLIPPER_CONFIG=${PRINTER_DATA}/config
export SYSTEMD=/etc/systemd/system
export KLIPPER_LOGS=${PRINTER_DATA}/logs
source ${KIAUH_SRCDIR}/scripts/utilities.sh
source ${KIAUH_SRCDIR}/scripts/mjpg-streamer.sh
install_mjpg-streamer

djObsidian, 2022-12-31:
@Creo2005 what is this? I should paste this into console? Or in kiauh.sh?

olivierchaput, 2023-01-11:
Hey @djObsidian, paste it into the console. Worked flawlessly for me! Thanks @Creo2005

JDPS, 2023-01-16:
Genius... simple and functional. Thanks.

SimonSolar2C, 2023-08-15:
Tried the above commands - was looking good but then got stuck.
Anyone know how to fix this?

###### Initializing MJPG-Streamer installation ...

###### Checking for the following dependencies:
● git
● cmake
● build-essential
● imagemagick
● libv4l-dev
● ffmpeg
● libjpeg62-turbo-dev

###### Installing the following dependencies:
● imagemagick
● libv4l-dev

-bash: : No such file or directory