dw-0/kiauh #76: klipper system service check issue on octopi

aschor, 2021-03-29:
Hi,

thanks for your hard work.

I tried to use kiauh on my actual system, that was (is) an octopi instance.

And sadly, my journey fails at the first step : installing moonraker. Because, as can be seen [here](https://github.com/KevinOConnor/klipper/blob/00706dec21446b6e4a869dced320c931bbc08d79/scripts/install-octopi.sh#L47) , klipper installs in /etc/init.d on octopi.

But as can be seen [here](https://github.com/th33xitus/kiauh/blob/67f6a2c59945353fd940d968587dfc43ae3a2476/scripts/install_moonraker.sh#L53) , kiauh searches for systemd unit to find out if klipper is installed, and hence fails to install moonraker with message " Klipper service not found, please install Klipper first!" .... but it's here, it's juste that kiauh does not search in the right place :/

th33xitus, 2021-03-29:
Correct. And that's intended. I made the decision to support systemd services only.

Removing Klipper and re-installing it with KIAUH will solve your problem.

aschor, 2021-03-29:
ok, thanks.

I've got 1 step further. Now I'm facing this message : 

/=======================================================\
| Possibly disruptive/incompatible services found!      |
|-------------------------------------------------------|
| ● haproxy                                             |
|-------------------------------------------------------|
| Having those packages installed can lead to unwanted  |
| behaviour. It is recommend to remove those packages.  |
|                                                       |
| 1) Remove packages (recommend)                        |
| 2) Disable only (may cause issues)                    |
| 3) Skip this step (not recommended)                   |
\=======================================================/

I'm not sure what to do to keep my octoprint instance working in case not everything goes well with moonraker (namely I have automations running between octoprint / mqtt / my home automation and I can't loose that until I've fully migrated to moonraker)

do you have any hints please ?  

(my setup was initially an octopi, not sure what version, that I had to apt dist-upgrade to latest buster to have a working ssd boot ; during the migration to buster I already faced obscure warnings about haproxy conf, and I kept the conf "as is" because I thought that it had been installed and configured to have octoprint work).

aschor, 2021-03-29:
reading the install code, I've got a lot of homework to do, and first of all : check that all my (few) addons work with python 3 ... I'll close this issue, thanks for the support

closed 2021-03-29 by aschor

th33xitus, 2021-03-29:
I suggest you use a second SD card and keep your working OctoPrint setup save if you can't afford to scrap that yet. :)