dw-0/kiauh #78: Default Pause / Resume Script

cfelicio, 2021-04-08:
Hi There,

I'm trying to adjust the default pause / resume script to account for the fact it's hard to change the filament on my printer with the stepper engaged on the extruder. I have the following script working on Octoprint:

PAUSE
{% if pause_position.x is not none %}
; relative XYZE
G91
M83
; retract filament, move Z slightly upwards
G1 Z+5 E-1 F4500 
; absolute XYZE
M82
G90
; move to a safe rest position, adjust as necessary
G1 X20 Y20
SET_STEPPER_ENABLE STEPPER=extruder ENABLE=0
{% endif %}

RESUME
{% if pause_position.x is not none %}
; relative extruder
M83
; prime nozzle
G1 E-1 F4500
G1 E1 F4500
G1 E1 F4500
; absolute XYZ
G90
M82 ; extruder relative mode
; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500
; reset to feed rate before pause if available
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}
G90

What is the easiest way to adjust the default script on Kiauh to include the "SET_STEPPER_ENABLE STEPPER=extruder ENABLE=0"? 

Also, I noticed the printer will park the extruder close to max X/Y. What is the easiest way to park it closer to min X/Y instead? In my script, I park around X20 Y20, to not touch the endstops, not sure how to modify to work in a similar way.

Thanks!

cfelicio, 2021-04-08:
here is what I'm planning to try so far:

#########################################################################################
# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  AUTOCREATED WITH KIAUH  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ #
#########################################################################################
# Those are the recommended macros and config entries if you use Mainsail or Fluidd!    #
# Feel free to edit or delete those macros if you already have them defined elsewhere!  #
#########################################################################################

[pause_resume]

[display_status]

[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
	G91
	G1 Z10 E-5 F4500
    TURN_OFF_HEATERS
    CLEAR_PAUSE
    SDCARD_RESET_FILE
    BASE_CANCEL_PRINT

[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
# change this if you need more or less extrusion
variable_extrude: 1.0
gcode:
    ##### read E from pause macro #####
    {% set E = printer["gcode_macro PAUSE"].extrude|float %}
    ##### set park positon for x and y #####
    # default is your max posion from your printer.cfg
    {% set x_park = printer.toolhead.axis_minimum.x|float + 10.0 %}
    {% set y_park = printer.toolhead.axis_minimum.y|float + 10.0 %}
    ##### calculate save lift position #####
    {% set max_z = printer.toolhead.axis_maximum.z|float %}
    {% set act_z = printer.toolhead.position.z|float %}
    {% if act_z < (max_z - 2.0) %}
        {% set z_safe = 2.0 %}
    {% else %}
        {% set z_safe = max_z - act_z %}
    {% endif %}
    ##### end of definitions #####
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{E} F2100
    G1 Z{z_safe} F900
    G90
    G1 X{x_park} Y{y_park} F6000
	SET_STEPPER_ENABLE STEPPER=extruder ENABLE=0

[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
    ##### read E from pause macro #####
    {% set E = printer["gcode_macro PAUSE"].extrude|float %}
    ##### end of definitions #####
    G91
    G1 E{E} F2100
    RESTORE_GCODE_STATE NAME=PAUSE_state
    BASE_RESUME

#########################################################################################
#########################################################################################

th33xitus, 2021-04-08:
Just edit them to your liking and how you need them. I am sorry but i don't feel responsible for checking and validating them, so that is up to you. I just provide those example macros.

cfelicio, 2021-04-08:
Yup, for sure. I just tested the above example I posted and it seems to work.

closed 2021-04-08 by cfelicio