dw-0/kiauh #80: Issue installing Klipper

rlesac, 2021-04-12:
I ran into this error:

`Apr 12 19:53:48 bpi-m2z systemd[338]: klipper.service: Failed to execute command: No such file or directory
Apr 12 19:53:48 bpi-m2z systemd[338]: klipper.service: Failed at step EXEC spawning /home/aramaki/klippy-env/bin/python: No such file or directory
Apr 12 19:53:48 bpi-m2z systemd[1]: klipper.service: Main process exited, code=exited, status=203/EXEC
Apr 12 19:53:48 bpi-m2z systemd[1]: klipper.service: Failed with result 'exit-code'.
`

```
###### Initializing Klipper installation ...

###### How many Klipper instances do you want to set up?: 1

###### Create 1 single instance? (Y/n): y
###### > Yes

###### Creating 1 Klipper instances ...

###### Downloading Klipper ...
Cloning into 'klipper'...
remote: Enumerating objects: 53, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 24605 (delta 31), reused 28 (delta 13), pack-reused 24552
Receiving objects: 100% (24605/24605), 16.71 MiB | 411.00 KiB/s, done.
Resolving deltas: 100% (18594/18594), done.
Updating files: 100% (1318/1318), done.

###### Download complete!

###### Installing dependencies ...

###### Running apt-get update...
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-doc' for regex 'libusb-1.0'
Package python-virtualenv is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-virtualenv' has no installation candidate

###### Installing python virtual environment...
/home/aramaki/kiauh/scripts/install_klipper.sh: line 72: virtualenv: command not found
/home/aramaki/kiauh/scripts/install_klipper.sh: line 74: /home/aramaki/klippy-env/bin/pip: No such file or directory

###### Creating Klipper Service  ...
Created symlink /etc/systemd/system/multi-user.target.wants/klipper.service → /etc/systemd/system/klipper.service.
>>>>>> Single Klipper instance created!

###### Launching Klipper instance ...

```

th33xitus, 2021-04-12:
Looks like you are using an unsupported base OS which is missing a python-virtualenv installation candidate.
Nothing i can do about it... you have to install that package somehow yourself or switch to a supported distro.
Raspberry Pi OS for example...

darkpoetry, 2021-04-19:
Debian 11 Bullseye will go stable soon, but Kiauh does not support it as of this posting. Bullseye does not have python-virtualenv.

So you need to install these packages first, it should work, at least per my experience.

virtualenv python2 python2-dev libffi-dev

I also installed, as troubleshooting, a number of other packages which I don't think impacted this but I'll list here in case they did: 
python3-cffi python3-openssl

closed 2021-05-05 by th33xitus