dw-0/kiauh #87: Multiple webcamd?

nibblerrick, 2021-05-27:
Hi,

thank you very much for this handy piece oft software!

I wanted to ask if you could consider installing multiple webcamd's with the script?
Doing manually what the script does to create another instance works (with adjusted filenames, of course) so I wondered if it could be achieved with the installer.

If you think this isn't something which belongs in your script feel free to close the issue right away.

Thanks!

th33xitus, 2021-05-27:
There is something in the planning. I have no ETA yet but yes, soon you should be able to run multiple webcams :)

nibblerrick, 2021-05-30:
That sounds awesome, thanks!

th33xitus, 2021-07-10:
KIAUH now uses the same NGINX configurations when installing Moonraker and Mainsail/Fluidd as they are used in MainsailOS or FluiddPi. That means it's possible to have up to 4 webcams at the same time. Those are configured each with their own webcam.txt file (webcam1.txt, webcam2.txt, ... ) in the klipper_config directory.

That way all webcams share the same webcamd service, which is pretty nice :)

closed 2021-07-10 by th33xitus

leandrotoledo, 2021-09-09:
Sorry to revive this one, but are there any additional instructions on how to setup multiple webcams?

I have two USB webcams in a multiple Moonraker/Klipper setup but only one instance of webcamd is spawned.

apphead, 2022-11-06:
In /home/pi/kiauh/resources/mjpg-streamer, I put webcam.txt and webcam2.txt and configured two cameras in settings. However, the config of webcam2.txt seems to never get picked up, resulting in the same camera displayed twice. Any ideas?

apphead, 2022-11-11:
Stupid me. I tried changing the port in (mainsail) camera settings, when all I needed to do is to squeeze a 2 in as in 
/webcam2?action=
 :laughing: :laughing: :laughing: :laughing:
[image]