dw-0/kiauh #91: KIAUH detects a manual Klipper install; subsequent moonraker install via kiauh fails by not finding klipper service

jim-sokoloff, 2021-06-04:
starting with an OctoPi based image, I git cloned 5d11cd212a5592c4348c42983586f91ee0209d67, ran kiahu.sh

main menu looks like this: (detects the klipper install)
```
/=======================================================\ 
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/
/=======================================================\ 
|     ~~~~~~~~~~~~~~~ [ Main Menu ] ~~~~~~~~~~~~~~~     |
|-------------------------------------------------------|
|  0) [Upload Log]     |       Klipper: Installed: 1    |
|                      |        Branch: master          |
|  1) [Install]        |                                |
|  2) [Update]         |     Moonraker: Not installed!  |
|  3) [Remove]         |                                |
|  4) [Advanced]       |      Mainsail: Not installed!  |
|  5) [Backup]         |        Fluidd: Not installed!  |
|                      | KlipperScreen: Not installed!  |
|  6) [Settings]       |                                |
|                      |          DWC2: Not installed!  |
|  v3.0.0-55           |     Octoprint: Installed: 1    |
|-------------------------------------------------------|
|                        Q) Quit                        | 
\=======================================================/
```

choosing 1,2 (install,  moonraker), yields:
```
###### Initializing Moonraker installation ...

###### Your Python 3 version is: Python 3.7.3
>>>>>> Python version ok!

#########################################################
 Klipper service not found, please install Klipper first! 
#########################################################
```

I'm going to go manually install moonraker, so I don't consider myself in any way blocked here, but I did want to pass along the observation as I tried this useful utility.
Once I've done that, I intend to come back and update this issue with how the Mainsail and Fluidd installs went.

th33xitus, 2021-06-04:
Uninstall Klipper with KIAUH and re-install it. After that you can install Moonraker with KIAUH too.

jim-sokoloff, 2021-06-04:
After a manual moonraker installer, kiauh installed mainsail without issue.
After the kiauh install of mainsail, kiauh installed fluidd without issue.

It was so easy, I'm going to make sure I've got everything backed up and then do as you suggest. (Meaning, this issue can be closed as "DONT_NEED_TO_FIX" :)

closed 2021-06-04 by jim-sokoloff