dw-0/kiauh #92: Cant find /etc/default/klipper

kolossboss, 2021-06-12:
Hi,

I'm running a manuell installation of klipper and Repetier Server (similar to octoprint) for a while now. Worked pretty good.

Today I tested the kiauh script. I needed to uninstall and reinstall klipper go get detected by kiauh.

My problem is. I need to edit the file at /etc/default/klipper to get Repetier Server working with klipper (permissions problem).
But I cant find this file anymore. Did the location changed?

Thanks a lot.

th33xitus, 2021-06-12:
KIAUH doesn't make use of sysVinit service files. It installs every service as systemd service.
You find the new klipper service in `/etc/systemd/system/klipper.service`.

kolossboss, 2021-06-13:
Hi,
thanks for reply.
 I changed to user to repetierserver, but the RS still cant connect to the printer. Is anything else changed? Maybe the virtual Printer Port?
RS wants to connect to /tmp/printer

Thanks for helping.

th33xitus, 2021-06-13:
No, the virtual printer is still the same.

You can do a quick self-check:
`ls /tmp/printer`
This command should return a result.

kolossboss, 2021-06-13:
Ok thanks, that works.
I also restartet the Pi and now I can connect with Repetier Server. 
But Mainsail and Octoprint did not work anymore....permissions error. Probably because I changed the user.
Do you have an idea how I can get RS server so work, without changing the user from "pi" to repetierserver"?

Thanks a lot!!

th33xitus, 2021-06-13:
Im sorry i am not familiar with RepetierServer, i never used it. I think you have to make use of Google to maybe find a solution for that, sorry.

I'll close this issue as it seems to be solved.

closed 2021-06-13 by th33xitus