dw-0/kiauh #94: DietPi and missing /var/log/nginx

beikeland, 2021-06-16:
On DietPi installation of klipper, moonraker and mainsail seems to succeed (after adressing #43 ) but nginx does not start because /var/log/nginx/access.log is missing, as a result of /var/log/nginx/ not existing.

Presumably due to the log to ram thing? I just did `sudo mkdir /var/log/nginx` and will perhaps look closer at it at a later time when i'm more familiar with DietPi.

th33xitus, 2021-06-16:
I would need to look closer into that myself too. I never used DietPi myself and for now don't officially support it with KIAUH. There just weren't any complains so far other than the dbus dependency which i added and should have worked.

Maybe i need to install DietPi myself on a Pi if i find some time for it to make the needed changes. But in general i always recommend to rather use either a RaspberryPi OS, Debian or Ubuntu 18 image.

th33xitus, 2021-06-18:
Regarding my answer in #43 i couldn't reproduce the issue with NGINX too.
I was able to install Klipper and Moonraker without a problem. Installing a webinterface and accessing it worked also. NGINX was running therefor.

closed 2021-06-29 by th33xitus