dw-0/kiauh #97: New install of kiauh doesn't create a functional klipper.bin for SKR1.3

LoganFraser, 2021-07-24:
I recently used kiauh to install klipper, mainsail and fluidd. I got an error connecting to my skr1.3 due to an older firmware, but had issues generating a functional klipper.bin file for it, getting usb connection fail messages on rpi bootup. *I did not manually install make until after I tried to generate the klipper.bin file manually*.  Is it possible there's a missing dependency for make in the kiauh installation process? After following the identical settings I was using in kiauh to make the firmware manually, I was able to connect to my skr1.3 again.

th33xitus, 2021-07-25:
Im not sure if i understand correct.
KIAUH doesn't create a klipper.bin file automatically just by installing/cloning it.

The user always has to go into the advanced menu and from there choose the appropriate option to either build only or build and flash the firmware.

LoganFraser, 2021-07-25:
Process followed to create a non-functional firmware using kiauh: 
Install kiauh using instructions from download page on a clean install of Raspbian buster. 
Install Klipper, Mainsail and fluidd via script. 
Create firmware via script, (4 then 3) using the following options: 

                                     Klipper Firmware Configuration
[ ] Enable extra low-level configuration options
    Micro-controller Architecture (LPC176x (Smoothieboard))  --->
    Processor model (lpc1768 (100 MHz))  --->
[*] Target board uses Smoothieware bootloader
    Communication interface (USB)  --->

rename the generated klipper.bin file to firmware.bin, copy to SD card. 
power off rpi and skr1.3
move SD card to skr1.3, restore power, wait 90 seconds, remove power.
remove SD card. 

On reboot, the Pi shows many USB connection failure messages, the SD card firmware.bin has been renamed firmware.cur indicating a successful flash of a non-functional firmware. 

So either the version of make used in the script is creating an issue, or the settings being passed from kiauh to make are not correct.

th33xitus, 2021-07-25:
> So either the version of make used in the script is creating an issue
KIAUH uses the same make which is used when you build the firmware by hand.

> or the settings being passed from kiauh to make are not correct.
For that to check, i need to know HOW you build the firmware by hand and which commands you exactly used.

LoganFraser, 2021-07-25:
sudo apt install make
make menuconfig
make clean
make

LoganFraser, 2021-07-25:
with the same options listed for firmware configuration: 
[ ] Enable extra low-level configuration options
    Micro-controller Architecture (LPC176x (Smoothieboard))  --->
    Processor model (lpc1768 (100 MHz))  --->
[*] Target board uses Smoothieware bootloader
    Communication interface (USB)  --->

th33xitus, 2021-07-25:
I see. I'll add an additional dependency check for `make` to the script before calling the make commands.
Hope that will help.

commit 39219c10 (2021-07-25): script: add additional dependency check - should fix #97

Before starting the firmware building process, check for dependencies again.
closed 2021-07-25 by th33xitus