dw-0/kiauh #98: cannot restart service after update

tjengbudi, 2021-08-02:
hello
i try the new kiauh , i used v2 before. and i tried version 3 now
the installation process do not have any problem
but the main problem is when fluid update the klipper, it cannot restart the service. in v2 it can without any problem

tjengbudi, 2021-08-02:
somehow sudo_fix.sh by moonraker is error. copy the file manually

and then i change the %mnrksudo (unknown variable) with my user manually. 

it works. 

the sudo_fix.sh cannot run at root even with command "sudo ./sudo_fix.sh" however in the code there is visudo command that must used root privilege. so i think kiauh must help with this

KwadFan, 2021-08-02:
Hello, could you please describe what exatly the Error should be? Can you reproduce this? If you could, pls post exatly the messages thrown out on your console/putty.

%mnrakrsudo isnt a Variable it is a group name. And you shouldnt replace this!
The only way this script fails is either your username isnt "pi" or you dont have a RaspberryOS based Distribution.

Cheers

tjengbudi, 2021-08-03:
hello,
sorry not to mention before. i am not using raspberry pi. i used debian armbian os. and the user is not pi. so it is known problem ?
i see that in the readme it tell that run in armbian. so with my trial i found this error.

th33xitus, 2021-08-03:
There were only reports that KIAUH works on Armbian, which are mentioned in the readme. There is no guarantee for it though.
It is unlikely that KIAUH V2 worked as you said and suddenly V3 doesn't.
Nothing has changed between V2 and V3 in this regard. KIAUH never made any use of the mentioned sudo_fix.sh bundled in Moonraker.

It's a permission issue KIAUH can't adress or fix automatically without user intervention as it seems. I can't fix anything and the issue seems to be invalid in the first place. I will close this issue therefore.

closed 2021-08-03 by th33xitus