dw-0/kiauh #99: Adding PrettyGCode to the installer

Kragrathea, 2021-08-10:
Hi! I am getting ready to release a new GCode visualizer called PrettyGCode (see below). It used to be an OctoPrint plugin but now I am porting it to Klipper/moonraker. It needs to be installed on the Raspberry Pi similar to how Fluidd is installed.  It is a single Nginx website with no other dependencies other than Moonraker for the API. 

Using Kiauh for distribution and updating would be idea, since I plan to eventually support all the different UIs (Fluidd,OctoPrint, Duet, Mainsail).  Would you be interested in collaborating on adding PrettyGCode into Kiauh? 
[PrettyGCode for Klipper](https://github.com/Kragrathea/pgcode)
[PrettyGCode for OctoPrint](https://github.com/Kragrathea/OctoPrint-PrettyGCode)

th33xitus, 2021-08-10:
Hey. I just added an installer for PrettyGCode with https://github.com/th33xitus/kiauh/commit/05842f8e1d2bcb4d03bf7d62467037a3a48602f9 😃

I made the port it runs on customizable for the user (it will default to 7136 though). 
I think that can be usefull to some people.

Kragrathea, 2021-08-10:
Wow! That was amazingly fast. :-) I'll test it as soon as I get home.

Kragrathea, 2021-08-10:
Works Great! I even tested an update.  Thanks so much th33xitus. Feel free to close this, if any problems come up I will create a new issue.

closed 2021-08-11 by th33xitus