dw-0/kiauh #100: Remove Mainsail or Fluidd needs cleanup

ssackett, 2021-08-14:
If you install both Mainsail and Fluidd and assign Fluidd to port 8080 and subsequently remove Fluidd two directories remain in the nginx config and can cause nginx to fail on startup if port 8080 has been assigned to a different listener such as a webcam.  Removing the directories fixes the problem.

The directories are:       
 /etc/nginx/sites-available/fluidd
/etc/nginx/sites-enabled/fluidd

th33xitus, 2021-08-14:
I just tried to reproduce the issue you mention. Without success.
It doesn't really matter which port you assign to Fluidd upon installation in this scenario.
If you remove it, usually the NGINX configuration files get deleted too: 
https://github.com/th33xitus/kiauh/blob/master/scripts/remove.sh#L289
https://github.com/th33xitus/kiauh/blob/master/scripts/remove.sh#L295

The only thing i see there is, that there is no --force or -f parameter used. But i am unsure if this would make any difference.
Can you reproduce this error every time?

ssackett, 2021-08-17:
I haven’t tried to reproduce the problem.  It happened while configuring a Voron and was one of many associated problems I had to track down.  I can’t remember the exact steps that I took but I think that I probably installed Mainsail and Fluidd manually without Kiauh and assigned 8080 to Fluidd at that time.  I do know that subsequently I used Kiauh to uninstall Fluidd to free up 8080 for a webcam and after rebooting the webcam worked but the Nginx service failed at startup because a port it was trying to use (8080) was already in use.  After I deleted the Fluidd scripts from the Nginx config the problem went a way.   Perhaps Kiauh didn’t know how to uninstall Fluidd because it hadn’t been used to install it.

th33xitus, 2021-08-19:
Okay, in that case im sorry that i have to close that issue now. But there is no way for me to try and find an issue that possibly doesn't exist right now. If you encounter the issue again, feel free to bump up this topic again and i will take a closer look again.

ssackett, 2021-08-19:
Kiauh is very convenient thanks for providing it.

ssackett, 2021-08-20:
I did a test on my RPi to see which commands work for deleting files and folders.  The results are below with comments.
-----------------------------
pi@voron:~ $ mkdir not-empty                              # Create a directory

pi@voron:~ $ touch not-empty/one                      # Add three files inside it
pi@voron:~ $ touch not-empty/two
pi@voron:~ $ touch not-empty/three

pi@voron:~ $ rmdir not-empty/                             # Attempt to delete the directory using "rmdir"
rmdir: failed to remove 'not-empty/': Directory not empty

pi@voron:~ $ rm not-empty                                  # Attempt to delete the directory using "rm"         
rm: cannot remove 'not-empty': Is a directory

pi@voron:~ $ rm -f not-empty/                             # Attempt to delete the directory using "rm -f"
rm: cannot remove 'not-empty/': Is a directory

pi@voron:~ $ rm -Rf not-empty/                            # The only command of these which works is "rm -Rf" 
pi@voron:~ $                                                            # (remove recursive force)

th33xitus, 2021-08-20:
> pi@voron:~ $ rm -Rf not-empty/ # The only command of these which works is "rm -Rf"

This is true. But the script doesn't delete a directory. Those are single files which get deleted. Therefore no recursive deletion is needed and a simple `rm` should be enough. Usually even without the `-f` flag.

ssackett, 2021-08-20:
I have no argument on your points, this is just info if someone has the same problem.  The files or directories were not deleted for some reason when I removed Fluidd using Kiauh and I was able to remedy the problem by manually deleting /etc/nginx/sites-available/fluidd and /etc/nginx/sites-enabled/fluidd using rm -Rf.  Perhaps that was unnecessary but it worked,  I thought it would be useful to the community to know that.  As far as reporting it as a Kiauh bug that was just a mechanism to make the solution known.  I could have continued without reporting my experience because for me the problem is solved.

th33xitus, 2021-08-21:
All good, no worries. I just wanted to explain a bit more :)

closed 2021-08-21 by th33xitus

edwardtich1, 2023-08-06:
rm /etc/nginx/sites-available/fluidd
rm: cannot remove '/etc/nginx/sites-available/fluidd': No such file or directory

Aurasphere-au, 2023-12-23:
I installed the Mainsail PI image and as I use Fluidd, installed Kiauh, removed mainsail and mainsail config, then installed fluidd. Fluidd accesses fine but these moonraker warnings now appear

[image]