dw-0/kiauh #101: [FR] Add timelapse component installer for Moonraker

thiccnugget, 2021-08-16:
Installing the timelapse component in mainsail for Moonraker is not the easiest thing for everyone. 

It would be nice to have an option in Kiauh to automatically install the timelapse component and to generate the configuration for the Moonraker.conf file and the macros for hyperlapse (timed timelapse) and standard timelapse (at layer change).

The end user can then choose to use either hyperlapse or the standard timelapse.

I'm talking about this
https://github.com/FrYakaTKoP/moonraker/blob/dev-timelapse/moonraker/components/timelapse.py

Whazor, 2022-01-12:
Timelapse component is currently in [a repository](https://github.com/mainsail-crew/moonraker-timelapse) of the mainsail crew. As the installation is similar to other klipper components I think it definitely makes sense to add it to kiauh.

CODeRUS, 2022-02-12:
kiauh have https://github.com/nlef/moonraker-telegram-bot which handles timelapse much better than mainsail extension

FrYakaTKoP, 2022-02-12:
@CODeRUS 
>>> which handles timelapse much better than mainsail extensio
Can you describe in more detail what you mean by that? I'm happy to improve the plugin. Btw. it's not a mainsail extension, it's a moonraker plugin/component. Mainsail only does integrate it into the Ui.

CODeRUS, 2022-02-13:
@FrYakaTKoP it uses opencv for image/video manipulations which is much much more faster and reliable

FrYakaTKoP, 2022-02-14:
@CODeRUS that's a fair point, but i took the wget/ffmpeg route, because it has other advantages. I may look into adding a option for opencv backend in future. But since the wget/ffmpeg variant runs pretty good in most cases and even on lowend hardware, i still don't get why refering this point as: 
> which handles timelapse much better than mainsail extension

Do you have any other suggestion, i may take on my journey improving the plugin?

CODeRUS, 2022-02-14:
@FrYakaTKoP thats okay for me, anyway, if using telegram bot no need to use any additional timelapse functionalities. no doubts can be useful for people who dont want telegram stuff on their printers. and as user i'd like to look at challenge making ultimate timelapse solution guys :))