dw-0/kiauh #102: Add 'Switch Fork' option

kylerw, 2021-08-20:
CR-10S Pro (V2) currently requires a Klipper fork by @Desuuuu (https://github.com/Desuuuu/klipper) to be able to utilize the touchscreen - would love to be able to fully use KIAUH and use this fork.

th33xitus, 2021-08-21:
Iirc using the desuuuu fork is only one part of the story, isn't it? Don't you need to flash the display with another firmware created by desuuuu too, to actually make use of it at all?

kylerw, 2021-08-22:
Yes, that’s correct. KIAUH wouldn’t be able to solve for the entire process but allowing installation of the modified Klipper would enable the rest of the features to be used more seamless

idontknowtom, 2021-09-12:
Since manual installation of Fluidd isn’t documented yet I figured out a way to get the Desuuuu fork of Klipper working with Fluidd by using KIAUH and modifying the “install_klipper.sh” script from “git clone $KLIPPER_REPO” to “git clone https://github.com/Desuuuu/klipper“. Then when you select options for the MCU firmware you select the touchscreen option and flash your mainboard. The printer config file needs this added:

[t5uid1]
firmware: dgus_reloaded

KIAUH should be able to do all this. Lastly, the user has to manually add the Desuuuu touchscreen firmware to a SD card and flash to the screen. This can’t be handled by KIAUH but having the option to install the Desuuuu fork and a link to the touchscreen firmware repo would be great.

th33xitus, 2021-09-12:
Do you need to use the master branch of his fork or does one usually then check out this branch?
https://github.com/Desuuuu/klipper/tree/dgus-display

If the master branch is used, it seems to be quite a lot more effort to make switching between the upstream and its fork possible without bigger hassles and possible issues.

kylerw, 2021-09-12:
This is great!

That is a newer branch with added customizations that was just released so not certain.

idontknowtom, 2021-09-12:
> Do you need to use the master branch of his fork or does one usually then check out this branch?
> https://github.com/Desuuuu/klipper/tree/dgus-display
> 
> If the master branch is used, it seems to be quite a lot more effort to make switching between the upstream and its fork possible without bigger hassles and possible issues. At least with the current mechanic i have in place to switch to the scurve-shaping branch for example. I would probably need to completely re-write that part of the script and handle stuff differently than it does now.

I have only used the Desuuuu master branch at this point. Makes sense that you have your switch fork script written in a certain way that may be difficult to implement this Desuuuu request. In the meantime, my simple replacement of part of line 88 in the "install_klipper.sh" script works fine to replace the master Klipper repo with the Desuuuu Klipper repo.

closed 2022-05-29 by th33xitus