dw-0/kiauh #103: Add an option to install Input Shaper dependencies & config

WSLAB3D, 2021-08-23:
- install dependencies:
pip install -v numpy
sudo apt update
sudo apt install python-numpy python-matplotlib

- Enable SPI
add "dtparam=spi=on" to /boot/config.txt

- Add config to printer.cfg
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None

[resonance_tester]
accel_chip: adxl345
probe_points:
    100,100,20  # an example

jbeima, 2021-11-14:
Some of this might not be needed.

If you are connecting the ADXL to an SPI port on boards like the Octopus, Octopus Pro, and SKR 1.2 Pro the [adxl345] block is configured for SPI communication.

Also you can't add the [mcu] block for the pi before the linux process is compiled and installed.

This maybe should be an option? Since there is more than one way to configure Input Shapping.

bjarchi, 2022-08-19:
Second that auto-installing all the deps for input shaping would be tricky, since how it's configured depends very much on one's specific setup -- e.g. which accel is used, where it's connected (and so whether the Pi MCU is needed), where that probe point is placed to be near the middle of the printer... 

I do have a FR open to add install of pi_mcu itself, ideally as a systemd unit instead of via init.d script. In parallel I'm looking into writing a systemd unit for it myself that will sequence correctly with Klipper.service; if I get that working then I'll submit a PR to Klipper.

DaVinci-10, 2023-07-02:
IMO..limiting the install for input shaper could simply just install numpy (the dependancies), but end the installer with a message to the user to add to printer.cfg etc,  I think would satisfy most people.

EricZimmerman, 2023-12-17:
look at doing Shake N Tune as the installer does all that for you, so it would be a simple one liner to install SnT