dw-0/kiauh #104: Add support for multiple camera instances when installing mjpg-streamer

WSLAB3D, 2021-08-23:
it would be good to be able to automate multiple camera instances to match klipper instances:
- create additional webcam*.txt for each camera - webcam.txt for first, webcam2.txt, webcam3.txt etc. for additionals.
- set port number for additional cameras: camera_http_options="-n -p <port number>" (webcam2 = 8081, webcam3 8082 etc)
- check that rpi camera is enabled (/boot/config.txt needs start_x=1, gpu_mem=128)

th33xitus, 2021-08-23:
> * check that rpi camera is enabled (/boot/config.txt needs start_x=1, gpu_mem=128)

Does this option affect usb cameras in any way? Or is it okay to just enable it no matter what?
Because you never know which kind of camera the user wants to run. I for myself only own a usb webcam, therefore i don't know if enabling the rpi camera somehow disabled the use of usb webcams?

WSLAB3D, 2021-08-23:
No, it doesn’t affect USB cameras, I use one of each. Perhaps to be safe it could be an option to enable/disable via the same menu?

jbeima, 2021-11-14:
I use a camera connected to the Pi CSI port, and as long as it is installed and setup in config.txt BEFORE you run Kiauh then mjpeg-streamer picks it up and sets up an instance for it.

Now if there was one of each I am not sure which one would take preference. Having it auto-configure more than one would be nice. :)

Right now it does see and configure an already enabled CSI camera.

tayanov, 2021-12-10:
I dont understand, now i can launch stream for 2 cameras? or not. If not i need it featufe too )))

bobafett81, 2022-02-21:
Im stuck on this atm

Kanrog, 2022-05-04:
Any progress on this? i have everything set up on my local server (Debian) running 2 printers and 2 cameras (2x C270's)
For the life of me i cant get the two cameras working simultaneously.  Seems like the problem is that KIAUH only installs one streamer and i dont have the knowlege to install a secoind one.

KIAUH is fantastic BTW :D

Skiddz, 2022-06-28:
Add one more to the "Any progress on this?" list.  I've been trying to get this working for weeks, but not getting anywhere.  I *can* make it happen on an OctoPi/Octoprint setup, but the implementation of mjpg-streamer on that platform is quite a bit different than the KIAUH setup.  I've tried several different ways to get another instance of mjpg-streamer running, but in all cases, mjpg-stereamer reports the 2nd device as always being "busy".

I can get either the PiCam *or* the USB cam working, but not both at the same time.  Looking at the installation script, I'm pretty sure I'm 'duplicating' all the right pieces (Like I do on the OctoPi/OctoPrint setups) but something isn't quite right.  Would love some help figuring this out.

ptwats, 2022-07-02:
I originally installed klipper with Kiauh and I think it is great. I did not try to install a second camera at the time but later put a second one from my Fluidd interface. If interested, I can share my webcam.txt and webcam2.txt for Fluidd. It may give you ideas for your setup.

Skiddz, 2022-07-06:
Are you running two USB cameras or a PiCam and a USB camera?  I'm running the latter and had it working great under OctoPi, but the KIAUH implementation of mjpg-streamer is quite different.  As I mention above, I can get one or the other camera working, but not both simultaneously.

Would I be correct in assuming you're switching between cameras and not running simultaneous streams?

waynemurray74, 2022-07-08:
I am running mainsail via my laptop, and it only picks up my built in front facing camera, not the usb connected camera. How do I make it use the usb camera??

ExPeacer, 2022-07-10:
I would say take a look to [Crowsnest](https://github.com/mainsail-crew/crowsnest) - making so easy to integrate two or more cameras.

Skiddz, 2022-07-15:
> I am running mainsail via my laptop, and it only picks up my built in front facing camera, not the usb connected camera. How do I make it use the usb camera??

Assuming you're running mjpg-streamer, you should just be able to edit webcam.txt.  Instead of "camera=auto" change it to "camera-usb" and then tweak whatever camera options you need to.

Once you've got the file edited, reboot the pi or restart webcamd:

sudo systemctl retsart webcamd

waynemurray74, 2022-07-15:
> Assuming you're running mjpg-streamer, you should just be able to edit webcam.txt. Instead of "camera=auto" change it to "camera-usb" and then tweak whatever camera options you need to.
>
> Once you've [...]

Wow, yeah ok, so I shut down my laptop out of frustration yesterday, trying to figure it out. After getting your response, I decided to grab it and try again. After powering it up, and loading mainsail, bam there's the webcam working. Apparently I just did not do the restart bit! *laughs in embarrassment* thank you!

closed 2023-02-03 by th33xitus