dw-0/kiauh #105: mjpg-streamer fix for Logitech C270

AnVa, 2021-08-24:
First of all, thank you so much for the work you have done!

I have a problem with Logitech C270 with Rpi 3B. I found a solution that fixes it

`sudo service webcamd restart`

maybe you should add this to auto start config of Rpi

Thank you!

th33xitus, 2021-08-25:
I don't quite understand why editing the rc.local is necessary for restarting the webcamd service. You can restart the webcamd service from within Mainsail or Fluidd iirc ?

AnVa, 2021-08-25:
Hi @th33xitus 
I can not. But a lot of people have the same problem
https://www.reddit.com/r/octoprint/comments/iwabfn/webcam_image_is_just_white_noise_logitech_c270/
https://www.reddit.com/r/octoprint/comments/ihivnv/logitech_c270_malfunction_white_noise_in_octoprint/

The problem is solved if i restart the service while loading Rpi

It is not issue for kiauh tool. I thought that this information might help someone, so I added it here 

thank you

closed 2021-09-11 by th33xitus