dw-0/kiauh #112: Submenu clarification to "back" from "quit"

The-Monkey-King, 2021-09-12:
Submenus reuse the quit_footer from general_ui.sh to quit the submenu and revert back to main menu. 

I proposed (PR #111 ) adding a back_footer to general_ui.sh and editing the submenus to use the word "back" to reflect the idea that you are going back to the main menu and not quitting the program. Quit is reserved for the main menu to quit the application. 

This is my very first PR so please excuse if I have done this wrong.

th33xitus, 2021-09-12:
Awesome!
I was actually about to do that change myself but now i can also just use your PR. I was told more than once, that "quit" wasn't the best choice there. I will have a look at the PR, test it and if it looks good, i will merge it soon. Thanks!

The-Monkey-King, 2021-09-12:
Ooops. I forgot to update upload_log.sh. Added to PR

th33xitus, 2021-09-12:
Thank you! Merged :)

closed 2021-09-12 by th33xitus