dw-0/kiauh #113: Instead of ports can kiauh user host names?

moebis, 2021-09-15:
In nginx and apache you can have it answer by host name instead of ip/port number. It would be nice to run all of the instances on port 80 but resolve by:

- http://fluidd.local
- http://mainsail.local
- http://octoprint.local

Is this possible?

th33xitus, 2021-09-21:
Hi,

i had the idea in the past and tried to find a solution to make that possible. But i wasn't successfull. I wasn't able to get a working configuration going even with the help of some people on the Klipper Discord. Also it's very likely that my knowledge on that topic just isn't good enough also. I made some researches if it's possible to somehow configure avahi in combination with NGINX to be able to use those URLs you wrote to access the different interfaces.

So is it possible? -> I don't know.

Do you have experience with it or even working configurations for NGINX?

moebis, 2021-09-25:
@th33xitus yes, I've run multiple websites on the same IP and port, just by defining different host names to listen on for each.

https://nginx.org/en/docs/http/server_names.html

Obviously each one pointing to a different root. I'm not familiar enough with these different front ends though. Do they all use nginx and something like php or node.js ....if they're all the same, it should be easy, especially with mainsail and Fluidd.

aeiounix, 2022-08-17:
If you have a DNS server that responds with the same IP address for each hostname requested, this works. That's beyond the scope of what's possible with a script. 
You can fake it with an entry in /etc/hosts on Windows and Linux. But it wouldn't work using simple DHCP. The server can know all that about itself but if there isn't something to tell the client it'd never know. 

My preference would be separate subdirectories. "Emotionally" it feels like it should just be an alteration to the various location clauses.

bjarchi, 2022-08-19:
@moebis Fluidd is a fork of mainsail made when Mainsail development had slowed (IIRC), so they are probably using the same or very similar plumbing on the backend. I'm not sure how octoprint relates, and this isn't exactly my area of expertise, but I'm happy to help with research and other ways that I'm able if you're still interested in this.

closed 2023-12-03 by dw-0