dw-0/kiauh #116: Add support for custom flash commands

ModMike, 2021-09-21:
Would it be possible to add a custom flash command for those of us who have non-compliant boards. I have a Fysetc Cheetah. I was thinking it could be a config option somewhere.

Thanks!

th33xitus, 2021-09-21:
Hi,

i think you are speaking of the flash command mentioned here, right?
https://github.com/KevinOConnor/klipper/blob/master/config/generic-fysetc-cheetah-v1.2.cfg#L8

Although i have an idea on how i could implement such a feature (for example asking the user for a one time input of the flash command which then gets saved to the kiauh.ini for later use) i have no possibility to test that feature then. It would be very experimental and i would relais 100% of user feedback on that. Which is no dealbreaker for me though... im just saying that it possibly won't work out of the box perfectly.

ModMike, 2021-09-21:
That would be PERFECT. Thanks!

And how do I donate?

th33xitus, 2021-09-21:
> And how do I donate?

😄Have a look here: https://github.com/th33xitus/kiauh/issues/115#issuecomment-924388549