dw-0/kiauh #120: Serial port question

ModMike, 2021-10-03:
I am trying to connect my Armbian Banana Pi M2 Zero to a Fystec Cheetah via UART3. The whole setup worked perfectly when using USB but I couldn't disable USB OTG on the BPI so I decided to go for serial connections.

I wired up the GND, TX and RX to the GPIO on the BPI pins 8, being sure to connect RX to TX and vice versa. The problem is I am confused at to how to configure everything. My BPI serial port is /dev/ttyS3 which I confirmed with Grep and get this result:

`$ dmesg | grep tty
[    0.000000] Kernel command line: root=UUID=e7e80086-a15f-4f2e-825d-9913c71ff19c rootwait rootfstype=ext4 console=ttyS0,115200 console=tty1 hdmi.audio=EDID:0 disp.screen0_output_mode=1920x1080p60 consoleblank=0 loglevel=1 ubootpart=bfca11dd-01 ubootsource=mmc usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   sunxi_ve_mem_reserve=0 sunxi_g2d_mem_reserve=0 sunxi_fb_mem_reserve=16 cgroup_enable=memory swapaccount=1
[    0.000353] printk: console [tty1] enabled
[    2.222269] printk: console [ttyS0] disabled
[    2.222365] 1c28000.serial: ttyS0 at MMIO 0x1c28000 (irq = 44, base_baud = 1500000) is a 16550A
[    2.222621] printk: console [ttyS0] enabled
[    2.223885] 1c28400.serial: ttyS1 at MMIO 0x1c28400 (irq = 45, base_baud = 1500000) is a 16550A
[    2.224070] serial serial0: tty port ttyS1 registered
[    2.224959] 1c28c00.serial: ttyS3 at MMIO 0x1c28c00 (irq = 46, base_baud = 1500000) is a 16550A
[    5.521136] systemd[1]: Created slice system-serial\x2dgetty.slice.
[    6.435591] systemd[1]: Found device /dev/ttyGS0.
[    7.852102] systemd[1]: Found device /dev/ttyS0.`

I configured Klipper as follows:

[mcu]
serial: /dev/ttyS3
restart_method: command

And I built the firmware with these options:

[ ] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32F103)  --->
    Bootloader offset (No bootloader)  --->
    Communication interface (Serial (on USART1 PA10/PA9))  --->

Questions:

1) I assume the MCU config in Klipper is for the board and the settings on the BPI are for the UART on the board, correct?
2) Are there other settings in kiauh that I need to be aware of or set?

Thank you

th33xitus, 2021-10-03:
Okay so you are using UART, currently i have hardcoded the UART port that gets used during flashing. The one that is hardcoded is `/dev/ttyAMA0` ... but i if you are saying you need to use `/dev/ttyS3` ... that will not work then of course.

I thought you need a special flash command for your board anyways? Referring to https://github.com/th33xitus/kiauh/issues/116#issue-1002449679
I haven't included such a feature yet but i will, i already have an idea for that.

I would make sure that you can flash the board manually over that serial port without kiauh. If you can confirm it working then, that would be helpfull to know. Because right now i can't really help troubleshooting without knowing that flashing that specific board over UART is usually possible.

ModMike, 2021-10-03:
Before I needed to go uart, I tested everything and it was working a treat through USB. Then I enabled UART 3 on the PI using Armbian Config made the change in the printer cfg MCU section and tried to flash:

stm32flash -w out/klipper.bin -b 115200 -v -i rts,-dtr,dtr /dev/ttyS3

but it give me a failed to nit device error