dw-0/kiauh #134: Fluidd update overrides gcode_shell_command plugin

plloppii, 2021-10-31:
I ran into this issue today, where after updating klipper through fluidd on my raspberry pi, the gcode_shell_command plugin was overwritten. 
I see in the install script, it places the python file within `klipper/klippy/extras/`
https://github.com/th33xitus/kiauh/blob/6b74c59d156007627a4f3ab38ccf17474ed67b08/scripts/functions.sh#L294

I wonder if there is a way to decouple this from the actual klipper file dir.

th33xitus, 2021-11-01:
Hi,

what do you mean by "overwritten"?
There is no such file in the whole Klipper repository which could overwrite it.
Also, the file doesn't get tracked by git at all. Unless you made it to do so, which would be a problem and shouldn't be done.
As long as the file doesn't get tracked by git, there should be no issues. 

I just updated Klipper through Mainsail (i updated Moonraker before that) and it worked out fine as usual. Moonraker does a simple git pull when updating. It's not cleaning up any untracked files before. Also i had a look at Moonraker and i saw no recent changes to the update manager. The last update to that module was last month. Since then updates work fine on my side at least.

So there must be something else that caused this.
Not even hard resets remove that file from the filesystem as you can see in the following console output. I reset Klipper 7 commits back, then did a git pull again. In between i did a git status each time to check repo status and you can see the file is always still there.

```
pi@zero2:~/klipper $ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        klippy/extras/gcode_shell_command.py

nothing added to commit but untracked files present (use "git add" to track)
pi@zero2:~/klipper $ git reset --hard HEAD~7
HEAD is now at 85d0ef97 stm32: Remove gcc -mfpu=fpv4-sp-d16 floating point option
pi@zero2:~/klipper $ git status
On branch master
Your branch is behind 'origin/master' by 7 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        klippy/extras/gcode_shell_command.py

nothing added to commit but untracked files present (use "git add" to track)
pi@zero2:~/klipper $ git pull
Updating 85d0ef97..d23c1b06
Fast-forward
 klippy/console.py         | 2 +-
 klippy/extras/palette2.py | 2 +-
 klippy/gcode.py           | 6 +++---
 klippy/serialhdl.py       | 6 +++---
 klippy/util.py            | 2 +-
 scripts/buildcommands.py  | 2 +-
 6 files changed, 10 insertions(+), 10 deletions(-)
pi@zero2:~/klipper $ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        klippy/extras/gcode_shell_command.py

nothing added to commit but untracked files present (use "git add" to track)
pi@zero2:~ $
```

plloppii, 2021-11-03:
oh interesting that does make alot of sense. Let me take a look again. It was weird because I clicked update (something I forgot if it was fluidd or klipper) and then fluidd started complaining about unrecognized gcode shell command in the config file.

closed 2021-11-22 by th33xitus