dw-0/kiauh #136: [Feature request] Option to update Python Installation(s)

LMS0815, 2021-11-08:
Please implement an option to (probe and) update outdated Python Installation(s) modules .

A script like this is working for me:
```
~/klippy-env/bin/pip list --outdated --format=freeze | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 ~/klippy-env/bin/pip install --upgrade
~/moonraker-env/bin/pip list --outdated --format=freeze | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 ~/moonraker-env/bin/pip install --upgrade
```
But I assume one need to take care of multiple instances.

Thank you.

th33xitus, 2021-11-08:
Can you explain why this change is needed?
Klipper and Moonraker both provide a requirements.txt where the exact python dependencys including their required version numbers are listed. So Klipper and Moonraker both require or want a specific version of each corresponding package installed. If those requirements change, KIAUH will detect that and will rebuild the virtualenv with the updated requirements.

> But I assume one need to take care of multiple instances.

All instances share the same virtualenv.

LMS0815, 2021-11-08:
Thank you for your reply. I keep on getting warning messages about outdated PIP.
As newer versions of pip contain bug fixes, and security improvements/vulnerability patches I like to be up to date. But I was not aware that Klipper and Moonraker maintain the requirements to a specific version because I assumed this is a minimum requirement.

th33xitus, 2021-11-08:
Do you mean a yellow warning popping up, that Python 2.7 is EOL during the creation of the venv? Thats normal and shouldn't be an issue once Klipper is ready for pure Python 3 systems.
In regards to PIP i think its best practice to upgrade pip globally anyways? If i am not wrong the command for that should be `pip install --upgrade pip`

Yes as you can see in the two following files, each package has a specific version assigned:
https://github.com/Klipper3d/klipper/blob/master/scripts/klippy-requirements.txt
https://github.com/Arksine/moonraker/blob/master/scripts/moonraker-requirements.txt
If newer versions than the ones specified in that textfile, i would assume they won't pick `==` but instead `>=`.

LMS0815, 2021-11-09:
In addition to the yellow EOL warning I got a warning about PIP being outdated as well on both  environments (2.7 and 3.x).

I got your point about the module versions and I assume the requirements are testes and known to be working.
Therefor I support your approach  maintaining the exact requirements. 

Are you using the -r option?
```
~/klippy-env/bin/pip install -r https://github.com/Klipper3d/klipper/blob/master/scripts/klippy-requirements.txt
~/moonraker-env/bin/pip install -r https://github.com/Arksine/moonraker/blob/master/scripts/moonraker-requirements.txt
```

But _PIP_ is not mentioned. Therefor I would be happy if you can implement some monitor and update routine.
The global update you mentioned above is "only" for the OS and dos not affect the virtual instances.

You  probably need to update any Python environment (e.g.):
```
~/klippy-env/bin/pip install --upgrade pip
~/moonraker-env/bin/pip install --upgrade pip
```

PS: I am not sure about _setuptools_ and _wheel_. I have no idea if they ate outdated or not.

closed 2022-05-29 by th33xitus